open62541/open62541 #4542: Local Directory Server not accepting connection after upgrading to v1.2.2

jackybek, 2021-07-19:
## Description

## Background Information / Reproduction Steps

Previously using v1.2.1 i have no issues with LDS accepting request from my OPCUA Server, or using Matrikon OPCUA Explorer Discovery function.

Using v1.2.2, my OPCUA server when connecting to LDS Server returned this error:
[2021-07-19 16:53:33.079 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2021-07-19 16:53:33.080 (UTC+0800)] info/userland      SV_StartOPCUAServer.c (LDS) : UA_LDS_connectUsername() : success
[2021-07-19 16:53:33.080 (UTC+0800)] error/server       **SV_StartOPCUAServer.c (LDS): registering to remote LDS server  : Fail at UA_Server_addPeriodicServerRegisterCallback. StatusCode BadInvalidState**

**Error message from Matrikon OPC UA Explorer:**

Find Server call for opc.tcp://192.168.1.44:4840 failed due to BadNotFound: A requested item was not found or a search operation ended without success.(0x803E0000)

Used CMake options:

BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_READELF:FILEPATH=/usr/bin/readelf
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_RPM:BOOL=OFF
CPACK_SOURCE_TBZ2:BOOL=ON
CPACK_SOURCE_TGZ:BOOL=ON
CPACK_SOURCE_TXZ:BOOL=ON
CPACK_SOURCE_TZ:BOOL=ON
CPACK_SOURCE_ZIP:BOOL=OFF
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdf
DVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUND
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUND
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUND
LIBWEBSOCKETS_INCLUDE_DIR:PATH=/usr/local/include
LIBWEBSOCKETS_LIBRARIES:FILEPATH=/usr/local/lib/libwebsockets.so
LIB_BPF:FILEPATH=LIB_BPF-NOTFOUND
LUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUND
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
MBEDCRYPTO_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libmbedcrypto.so
MBEDTLS_INCLUDE_DIRS:PATH=/usr/include
MBEDTLS_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libmbedtls.so
MBEDX509_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libmbedx509.so
MDNSD_LOGLEVEL:STRING=300
OPENSSL_CRYPTO_LIBRARY:FILEPATH=/usr/local/ssl/lib/libcrypto.so
OPENSSL_INCLUDE_DIR:PATH=/usr/local/ssl/include
OPENSSL_SSL_LIBRARY:FILEPATH=/usr/local/ssl/lib/libssl.so
PDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUND
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PKG_CONFIG_EXECUTABLE:FILEPATH=/usr/bin/pkg-config
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
SPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-build
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ARCHITECTURE:STRING=posix
UA_ARCH_ADD_FLAGS:STRING=
UA_ARCH_EXTRA_INCLUDES:STRING=
UA_ARCH_LINKER_FLAGS:STRING=
UA_ARCH_REMOVE_FLAGS:STRING=
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_DEBUG_FILE_LINE_INFO:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=ON
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OPENSSL
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MQTT_TLS:BOOL=OFF
UA_ENABLE_MQTT_TLS_OPENSSL:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_BUFMALLOC:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_ENCRYPTION:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=ON
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_PUBSUB_MONITORING:BOOL=ON
UA_ENABLE_PUBSUB_MQTT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=ON
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): v1.2.2-363-ge1f8fd3b
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ X] Critical issue

jackybek, 2021-07-20:
Solved by 
1. Changing to :: if (client->connection.state == UA_CONNECTIONSTATE_CLOSED) or
2. Changing to :: if (client->connection.state != UA_CONNECTIONSTATE_ESTABLISHED)

closed 2021-07-20 by jackybek

embbo, 2022-09-06:
@jackybek
have your tried with STM32 ?