open62541/open62541 #4546: UA_Client: How to retrieve the namespace of the UA_Server so as to update the nodes?

jackybek, 2021-07-20:
## Description
I am able to create a OPCServer with full namespace, assigned a namespaceindex at runtime using the following command:

                uaServer1 = UA_Server_newWithConfig(&config1);
                size_t namespaceIndex;
                //Add a new namespace to the server => Returns the index of the new namespace i.e. namespaceIndex
                UA_Int16 nsIdx_MKS = UA_Server_addNamespace(uaServer1, "virtualskies.com.sg/MKS/"); //"http://virtualskies.com.sg/MKS/"); UA_LOG_INFO(UA_Log_Stdout,
                UA_Server_getNamespaceByName(uaServer1, UA_STRING("virtualskies.com.sg/MKS/"), &namespaceIndex);
                UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "New Namespace added: <virtualskies.com.sg/MKS/>, nsIdx_MKS assigned is %d", nsIdx_MKS); // create nodes

In the above example, the namespaceindex generated at runtime is 2 (viewed using UAExpert), ns=2.

I now want to use a OPCUA Client instance to connect to this UAServer and update the address space.
However, this is no such function to 'getNamespaceByName' using a uaClient object.

Please advise how I can go about retrieving the Server namespace so that I can update the nodes with data using a UAClient.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-07-20:
I am not sure, but may your client can read the base node "NamespaceArray"; NodeId ns=0;i=2255.

https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.1/
NamespaceArray defines an array of namespace URIs. This Variable is also referred as namespace table. The indexes into the namespace table are referred to as NamespaceIndexes. NamespaceIndexes are used in NodeIds in OPC UA Services, rather than the longer namespace URI. Index 0 is reserved for the OPC UA namespace, and index 1 is reserved for the local Server. Clients may read the entire namespace table or they may read individual entries in the namespace table. The Server shall not modify or delete entries of the namespace table while any client has an open session to the Server, because clients may cache the namespace table. A Server may add entries to the namespace table even if clients are connected to the Server. It is recommended that Servers not change the indexes of the namespace table but only add entries, because the client may cache NodeIds using the indexes. Nevertheless, it might not always be possible for Servers to avoid changing indexes in the namespace table. Clients that cache NamespaceIndexes of NodeIds should always check when starting a session to verify that the cached NamespaceIndexes have not changed.

jackybek, 2021-07-22:
I found the function.. it is:

UA_StatusCode
UA_Client_NamespaceGetIndex(UA_Client *client, UA_String *namespaceUri,
                            UA_UInt16 *namespaceIndex)

jackybek, 2021-07-22:
tested successfully.

closed 2021-07-22 by jackybek