open62541/open62541 #4559: Open26541 build fails during linking with clang-8 and clang-10

ftpronk, 2021-07-26:
## Description

Compiling the branch tagged `1.2` with `clang-8` or `clang-10` fails on `Ubuntu 18.04`.

Clang-8 returns the following error:

```
[100%] Linking C shared library bin/libopen62541.so
clang: error: argument unused during compilation: '-Qn' [-Werror,-Wunused-command-line-argument]
CMakeFiles/open62541.dir/build.make:190: recipe for target 'bin/libopen62541.so.1.2.2' failed
make[2]: *** [bin/libopen62541.so.1.2.2] Error 1
CMakeFiles/Makefile2:523: recipe for target 'CMakeFiles/open62541.dir/all' failed
make[1]: *** [CMakeFiles/open62541.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

Clang-10 returns the following error:
```
[100%] Linking C shared library bin/libopen62541.so
clang: error: argument unused during compilation: '-fno-ident' [-Werror,-Wunused-command-line-argument]
CMakeFiles/open62541.dir/build.make:190: recipe for target 'bin/libopen62541.so.1.2.2' failed
make[2]: *** [bin/libopen62541.so.1.2.2] Error 1
CMakeFiles/Makefile2:523: recipe for target 'CMakeFiles/open62541.dir/all' failed
make[1]: *** [CMakeFiles/open62541.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

## Background Information / Reproduction Steps

1. Clone from GitHub, checkout the `1.2` branch, initialize the submodules and create a build directory
2. Compile with the following options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): `1.2`
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: `Ubuntu 18.04`
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

lazka, 2021-10-30:
Same issue when building with clang 13 on Windows:

```
clang: error: argument unused during compilation: '-fno-ident' [-Werror,-Wunused-command-line-argument]
ninja: build stopped: subcommand failed.
```

ftpronk, 2021-11-02:
> Hmm. I am surprised, because we do build with clang for the CI. We might need separate build flags for linker and compiler. As a workaround you can comment out the line for the "-Werror" option in the CMakeLists.txt.

I guess it comes down to some defaults enabled differently per OS / toolchain / CMake release, which would be hard to catch with CI.  Separating the build flags is in itself a good idea, and is made easier with modern CMake, but the refactoring can come with a bit of hair-pulling..  Well worth it in the end though, as, from experience, the global flags can have some "funny"/painful side effects..  Godspeed if you go down that road!

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The offending CMake flag was removed and is gone on all maintained branches (see d803a74), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier