open62541/open62541 #4561: Basic Data Type Tutorial Segmentation Fault

oerlikoncw, 2021-07-27:
## Description
I try to follow the tutorials from the documentation. When running the BasicDataHandling tutorial I get a segmentation fault on:

_UA_init(&rr, &UA_TYPES[UA_TYPES_READREQUEST]);_

However the next line (if commenting the previous one) works:

_UA_ReadRequest_init(&rr);_

## Background Information / Reproduction Steps

I have the open62541.c and open 62541.h file in the same folder as the c code file.
I build the application using:

```bash
gcc -std=c99 open62541.c basicDataHandling.c -o BasicDataHandling
./BasicDataHandling
```

I wonder if I do something wrong here...

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 1.2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system: Ubuntu 21.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-08-04:
Funny.
Those two lines are identical.

Please you run this with valgrind or in a debugger to get more information.

andreasebner, 2025-07-30:
As there's been no activity after @jpfrs comment, I'll close this issue

closed 2025-07-30 by andreasebner