open62541/open62541 #4563: Segmentation fault when adding a custom data type, following tutorial

JensMertelmeyer, 2021-07-28:
## Background Information
I am trying to follow the tutorial "[Working with variable data types](https://open62541.org/doc/1.2/tutorial_server_variabletype.html#working-with-variable-types)".

This is my code:
```cpp
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/log_stdout.h>

void addCustomType(UA_Server* server);

int main(int argc, char **argv) {
	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addCustomType(server);
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}
```

## Behaviour

1. The Variable Type node is added without errors
2. The server starts up flawlessly
3. Client (UaExpert) can connect and browse with no issues
4. As soon as the client tries to read the node `Types > VariableTypes > BaseVariableType > BaseDataVariableType > Thickness Profile`, my application panics!

```
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ffb301312de in memcmp () from vcruntime140.dll
```
and the following callstack:

```
vcruntime140.dll!memcmp
open62541.dll!open62541!UA_Array_copy
open62541.dll![Unknown/Just-In-Time compiled code]
open62541.dll!open62541!UA_copy
open62541.dll!open62541!__UA_Server_write
open62541.dll!open62541!__UA_Server_write
open62541.dll!open62541!UA_Server_call
open62541.dll!open62541!UA_Server_call
open62541.dll!open62541!UA_Server_findDataType
open62541.dll!open62541!UA_Server_call
open62541.dll!open62541!UA_Server_removeConnection
open62541.dll!open62541!UA_Server_removeConnection
open62541.dll!open62541!UA_Server_removeConnection
open62541.dll!open62541!UA_readNumberWithBase
open62541.dll!open62541!UA_readNumberWithBase
open62541.dll!open62541!UA_readNumberWithBase
open62541.dll!open62541!UA_Server_processBinaryMessage
open62541.dll!open62541!UA_DateTime_nowMonotonic
open62541.dll!open62541!UA_Server_run_iterate
open62541.dll!open62541!UA_Server_run
main(int argc, char ** argv) (program.cpp:14)
```

## Full reproduction
Here is the code for the above `addCustomType(..)`:

```cpp
void addCustomType(UA_Server* server) {
	const auto TYPE = UA_TYPES[UA_TYPES_FLOAT];
	const size_t COUNT = 360;
		
	// basic variable type attributes
	UA_VariableTypeAttributes vta = UA_VariableTypeAttributes_default;
	vta.dataType            = TYPE.typeId;
	vta.valueRank           = UA_VALUERANK_ONE_DIMENSION;
	vta.arrayDimensionsSize = 1;
	UA_UInt32 arrayDims[1]  = {360};
	vta.arrayDimensions     = arrayDims;
	vta.displayName         = UA_LOCALIZEDTEXT_ALLOC("en", "Thickness profile");
	vta.description         = UA_LOCALIZEDTEXT_ALLOC("en", "A list of usually 360 values, representing the thickness in microns");
				
	// default value
	UA_Float defaults[COUNT] = {};
	UA_Variant_setArray(&vta.value, defaults, COUNT, &TYPE);
	
	UA_NodeId newNodeId;
	auto ret = UA_Server_addVariableTypeNode(
		server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		UA_QUALIFIEDNAME_ALLOC(1, "ThicknessProfile"),
		UA_NODEID_NULL,
		vta,
		nullptr,
		&newNodeId
	);
	const auto statusText =  UA_StatusCode_name(ret);
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "addVariableTypeNode(..): 0x%x (%s)", ret, statusText);
}
```

## Idea
I suspect (but can't prove) that `UA_Server_findDataType` returns `NULL` and this causing problems later on.
Because when I call `UA_Server_findDataType`, it returns nothing although I believe it should:

```cpp
	const auto dataType = UA_Server_findDataType(server, &newNodeId);
	if(!dataType)
		UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "UA_Server_findDataType returned NULL");

```

## Checklist

Please provide the following information:

 - open62541 Version: 1.2
 - Other OPC UA SDKs used (client or server): Client: **UaExpert**
 - Operating system: Windows x64
 - Logs (with `UA_LOGLEVEL` set as low as necessary) attached ✔
 - ~~Wireshark network dump attached~~
 - Self-contained code example attached ✔

JensMertelmeyer, 2021-07-28:
I think I found the cause of the segmentation fault, the callstack already indicated it with `open62541!UA_Array_copy`:

I blindly copied the example
https://open62541.org/doc/1.2/tutorial_server_variabletype.html#working-with-variable-types
which has the following code:

```c
    /* a matching default value is required */
    UA_Double zero[2] = {0.0, 0.0};
    UA_Variant_setArray(&vtAttr.value, zero, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
```
The documentation of `UA_Variant_setArray` states
> Set the variant to an array that already resides in memory. The array takes
> on the lifecycle of the variant and is deleted with it.

I think in my example, the array was on the stack and when the server tried to provide a default value, it was trying to access that address.
Everything works as expected (no segfault) when I change my two lines from 
```cpp
	UA_Float defaults[COUNT] = {};
	UA_Variant_setArray(&vta.value, defaults, COUNT, &TYPE);
```
to
```cpp
	UA_Float *defaults = new UA_Float[360]{};
	UA_Variant_setArray(&vta.value, defaults, COUNT, &TYPE);
```

**Does that mean the documentation/example is wrong?**

jpfr, 2021-07-30:
You are adding a variable type, not a datatype. So finddatatype returns NULL.

jpfr, 2021-07-30:
Thr callstack is impossible.
UA_readNumberWithBase does not call the server.

Do you have debug symbols enabled?

JensMertelmeyer, 2021-07-30:
Thank you for looking into it. My DLL was probably not compiled to emit good debug output. I will see if I can manage to get it compiled it again with other settings, I found no binaries for 1.2 anywhere on GitHub or open62541.org.

> You are adding a variable type, not a datatype. So finddatatype returns NULL.
Yes, that makes sense. Sorry for bringing this up.

jpfr, 2021-08-04:
No biggie.
Just reopen if this remains problematic after further digging.

You can use cmake to generate a visual studio project for the dll if you need that.

closed 2021-08-04 by jpfr