open62541/open62541 #4568: Get node id from the nodecontext after successful node creation

Maku-chan, 2021-07-30:
#Description
I want to get the node id using the getnodeContext method. How can I get the node id if I call the function of this class to add nodes from a different file.

The below mentioned class is just an example.

class newdevice {
        public:
             void test() {
  
                       UA_Server_addObjectTypeNode(Internal_server, UA_NODEID_NULL,
			         UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
			         UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
			         UA_QUALIFIEDNAME(namespace_id, qualified_name), object_attr,
			         NULL, &deviceNodeType);
             }
}

jpfr, 2021-08-02:
The outNodeId is set automatically if you provide a pointer to a NodeId to be overwritten.

Given that adding is successful...

jpfr, 2021-08-04:
Reopen if required.

closed 2021-08-04 by jpfr