open62541/open62541 #4569: segmentation fault : v1.2.2-372-g6f21ecb4

jackybek, 2021-07-31:
## Description
if i compile my code with b**uild v1.2.2-363-ge1f8fd3b**, there is no issue.

## Background Information / Reproduction Steps
[2021-07-31 14:49:26.390 (UTC+0800)] info/userland      starting server...
[2021-07-31 14:49:26.390 (UTC+0800)] info/userland      StartOPCUAServer.c : after loading certificate and privateKey
[2021-07-31 14:49:26.392 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2021-07-31 14:49:26.393 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
[2021-07-31 14:49:26.393 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2021-07-31 14:49:26.394 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2021-07-31 14:49:26.394 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-07-31 14:49:26.394 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
hostname.data (ip) = 192.168.1.33
[2021-07-31 14:49:26.394 (UTC+0800)] info/userland      StartOPCUAServer.c : after making changes to config1
[2021-07-31 14:49:26.394 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-07-31 14:49:26.394 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
[2021-07-31 14:49:26.394 (UTC+0800)] info/userland      adding 2 user credentials to OPCUA server ...

[2021-07-31 14:49:26.394 (UTC+0800)] info/userland      starting OPCUA server ...

[2021-07-31 14:49:26.623 (UTC+0800)] info/userland      New Namespace added: <virtualskies.com.sg/MKS/>, nsIdx_MKS assigned is 2
======================In UA Nodes setup=======================
[2021-07-31 14:49:26.623 (UTC+0800)] info/userland      Root Node created : Airgard
[2021-07-31 14:49:26.623 (UTC+0800)] info/userland      Attribute Variable created: Airgard->VersionType->SoftwareVersion 10001

Program received signal SIGSEGV, Segmentation fault.
**0xb6fba428 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so**
(gdb) backtrace
#0  0xb6fba428 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
#1  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Used CMake options:

-- Cache values
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/gcc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_READELF:FILEPATH=/usr/bin/readelf
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_RPM:BOOL=OFF
CPACK_SOURCE_TBZ2:BOOL=ON
CPACK_SOURCE_TGZ:BOOL=ON
CPACK_SOURCE_TXZ:BOOL=ON
CPACK_SOURCE_TZ:BOOL=ON
CPACK_SOURCE_ZIP:BOOL=OFF
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdf
DVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUND
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUND
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUND
LUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUND
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
PDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUND
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
SPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-build
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ARCHITECTURE:STRING=posix
UA_ARCH_ADD_FLAGS:STRING=
UA_ARCH_EXTRA_INCLUDES:STRING=
UA_ARCH_LINKER_FLAGS:STRING=
UA_ARCH_REMOVE_FLAGS:STRING=
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_DEBUG_FILE_LINE_INFO:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MQTT_TLS:BOOL=OFF
UA_ENABLE_MQTT_TLS_OPENSSL:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_BUFMALLOC:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ENCRYPTION:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_PUBSUB_MONITORING:BOOL=OFF
UA_ENABLE_PUBSUB_MQTT:BOOL=OFF
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND
_Python3_EXECUTABLE:FILEPATH=/usr/bin/python3.7

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): v1.2.2-372-g6f21ecb4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jpfr, 2021-08-01:
Please enable debug symbols for a better backtrack.
We can fix much faster if the backtrack contains method names.

jackybek, 2021-08-02:
ok.. let me resolve the secure MQTT issue first, then I will switch to the latest version of the library.  in the meantime, can you take a look at the issue reported in open62541.org -> community mailing list ? 
Topic is : opcServer with MQTT TLS option

jackybek, 2021-08-02:
which option in cmake do you want me to turn on?

jackybek, 2021-08-02:
is it the UA_DEBUG and UA_DEBUG_FILE_LINE_INFO?

jackybek, 2021-08-05:
i reverted to v1.2.1-dirty version but the situation turned even worse.
Previously using v1.2.2XXX i can at least execute 10+ UA_addVariableNode().
Now with v1.2.1-dirty, after the first UA_addVariableNode() it crashed. It always crashed at this command when i use gdb to trace

/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so

I already perform a linux update prior: apt-get update and apt-get upgrade

***
[2021-08-05 14:23:25.488 (UTC+0800)] info/userland      In StartOPCUAServer thread: g_argc=6

[2021-08-05 14:23:25.488 (UTC+0800)] info/userland      starting server...
[2021-08-05 14:23:25.488 (UTC+0800)] info/userland      StartOPCUAServer.c : after loading certificate and privateKey
[2021-08-05 14:23:25.491 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2021-08-05 14:23:25.492 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
[2021-08-05 14:23:25.492 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2021-08-05 14:23:25.493 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2021-08-05 14:23:25.493 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-08-05 14:23:25.493 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
hostname.data (ip) = 192.168.1.33
[2021-08-05 14:23:25.493 (UTC+0800)] info/userland      StartOPCUAServer.c : after making changes to config1
[2021-08-05 14:23:25.493 (UTC+0800)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-08-05 14:23:25.493 (UTC+0800)] info/server        AccessControl: Anonymous login is enabled
[2021-08-05 14:23:25.493 (UTC+0800)] info/userland      adding 2 user credentials to OPCUA server ...

[2021-08-05 14:23:25.493 (UTC+0800)] info/userland      starting OPCUA server ...

[2021-08-05 14:23:25.733 (UTC+0800)] info/userland      New Namespace added: <virtualskies.com.sg/MKS/>, nsIdx_MKS assigned is 2
======================In UA Nodes setup=======================
[2021-08-05 14:23:25.733 (UTC+0800)] info/userland      Root Node created : Airgard
[2021-08-05 14:23:25.733 (UTC+0800)] info/userland      Attribute Variable created: Airgard->VersionType->SoftwareVersion 10001

**Program received signal SIGSEGV, Segmentation fault.
0xb6fb9bf0 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so**

jackybek, 2021-08-05:
I did some more digging and found the following interesting lead:

1.  Running on kunbus RevPi Core - no issue
uname -a
Linux RevPi-110 4.19.95-rt38-v7 #1 SMP PREEMPT RT Tue, 22 Jun 2021 14:13:31 +0000 armv7l GNU/Linux
Version of open62541:
**#define UA_OPEN62541_VERSION "v1.2.2-394-g5edb3a02-dirty"**

2. Running on a Rasberry pi 4 - segmentation fault at UA_addVariableNode()
uname -a
Linux OPCServer-33 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux
Version of open62541:
**#define UA_OPEN62541_VER_COMMIT "v1.2.1-dirty"**
I also get the same outcome if i use v1.2.2 variation

jackybek, 2021-08-06:
i did another test on another raspberry pi :
uname -a
Linux OPCServerUAT-112 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
#define UA_OPEN62541_VERSION "v1.2.2-363-ge1f8fd3b"

As expected, running open62541 v1.2.2 using an older version of **linux 5.10.17-v7+** runs perfectly.

So i conclude that open62541 v1.2.2 using a newer version of **linux 5.10.52-v7l+** will cause segmentation fault.

Please advise when will this be fixed

jackybek, 2021-08-22:
remove this when initialising new nodes using UA_Server_addVariableNode():

        //char SoftwareVersion[MAX_STRING_SIZE] = "open62541";
	//if (!UA_Variant_isEmpty(&vSoftwareVersionAttr.value))
	        //UA_Variant_setScalar(&vSoftwareVersionAttr.value, &SoftwareVersion, &UA_TYPES[UA_TYPES_STRING]);
        //else UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
        //        "Error UA_Variant_setScalar() : SoftwareVersion 10001");

closed 2021-08-22 by jackybek

jasonmnemonic, 2022-04-04:
Hi. 

I have a similar issue to the above and I am not sure why. My code is very simple (in principle) where I am web streaming in which video from the Pi Camera and Zero 2 W goes into a splitter then into my code through MMAL. My code takes the camera output and do stuff. 

I have been having similar crashes to the above where I have built my entire code with debugging symbols and stuff. Essentially, I am getting similar crash to the above i.e. `memcpy` and `libarmmem-v7l.so`.

My OS is `Linux MyPi 5.10.82-v7+ #1497 SMP Fri Dec 3 16:26:57 GMT 2021 armv7l GNU/Linux`

and my gdb output with backtrace is:
```
Thread 37 "vc.ril.video_en" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x684ed300 (LWP 2424)]
0x76fba44c in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so

(gdb) bt
#0  0x76fba44c in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
#1  0xfffffffe in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

OK, the weird thing is, most of my stuff are normally running OK i.e. sometimes for hours without crashing and sometimes it happens. It could also be my code by for the time being, it looks OK. The only thing I can roughly narrow down is the MMAL and my code interface.

As I said earlier, the video output goes through a splitter in which its second output gets the callback to copy the port buffer into my code's memory via memcpy.

```
mmal_buffer_header_mem_lock( buffer );
// myblob's image_buffer is same or greater than buffers.data
// at the start myblob is allocated to have enough buffer to copy e.g. 1080p has 1088 array length
// buffer is YUV so at 1080p, I am actually reading 1920x1088 size from buffer->data into myblob which is also the same size 
memcpy( myblob_->image_buffer, buffer->data, myblob_->image_buffer_size );  
mmal_buffer_header_mem_unlock( buffer );
```

In my head, I should not have segmentation fault but here I do... Any idea??

I can find some reference to https://forums.raspberrypi.com/viewtopic.php?t=44982&start=25 but I do not think I have incorrect sized array.

Help or advice please, thanks. :-)