open62541/open62541 #4573: Segmentation fault on sampling MonitoredItem

JohannesKauffmann, 2021-08-02:
## Description
Every two to five days, we get a random, non-reproducable segmentation fault. As the backtrace shows, the issue seems to occur during the sampling of a MonitoredItem. We have a client subscribed to this node. The minimum sampling and publishing interval is 200ms.

The MonitoredItem in question monitors an normal scalar value UA_Int32 variable node. From a core dump we gathered, the UA_Variant `src` in stack frame 8 has has `arraySize` of 0 and the `data` pointer > 0x01 (which indicates a scalar).

The backtrace also contains line numbers for our amalgamated build which should be reproducable with the cmake commands and git tag used.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake .. -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Debug -DUA_NAMESPACE_ZERO=REDUCED -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ARCH_REMOVE_FLAGS="-Werror" -DUA_ARCH_REMOVE_FLAGS="-fvisibility=hidden"
```

Used CFLAGS during compilation: `-O2 -pipe -g -feliminate-unused-debug-types -fvisibility=default -DNDEBUG=1`

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tag 1.2.2
 - [x] Other OPC UA SDKs used (client or server): UAExpert client 1.5.1 331
 - [x] Operating system: Poky Linux 2.6.1 for server, Windows 10 2004 for UAExpert
 - [x] GDB backtrace attached:
[github-bt.txt](https://github.com/open62541/open62541/files/6916989/github-bt.txt)

 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2021-08-02:
We’ll try to reproduce.
Do you access the server from parallel threads or is all single-threaded?

JohannesKauffmann, 2021-08-03:
The `UA_Server_run_iterate` function is called from it's own thread. Functions such as `UA_Server_writeValue`, `UA_Server_readValue` and `UA_Server_readDataType` are called from separate threads, yes.

jpfr, 2021-08-03:
Then please enable multithreading in the build flags.
I don't see that in the posted string.

jpfr, 2021-08-03:
It seems we should enable thread-safety by default....

JohannesKauffmann, 2021-08-03:
Alright, which value of UA_MULTITHREADED do you recommend? Is there any difference between a value of 100, 101 or 198?

jpfr, 2021-08-03:
Just 100. That adds locks to the server so it can be used from parallel threads.

JohannesKauffmann, 2021-08-04:
Thanks a lot, can't believe we missed this. Currently testing to see if it fixes the issue. Closing for now.

closed 2021-08-04 by JohannesKauffmann