open62541/open62541 #4574: 使用安全策略时，出现：验证安全性时出错

liutt1312, 2021-08-03:
使用安全策略时，出现：验证安全性时出错 这是什么导致的

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-08-16:
Affirmative.. in the latest patch release 1.2.2.xxxx, when I connect UAExpert to opcua server, I get BadSecurityPolicy rejected if I choose any of the following in UAExpert:
1. Basic256Sha256
2.  Aes128_Sha256_RsaOaep 
3. Aes256_Sha256_RsaPss 
4. None

Manishcimcon, 2021-08-18:
You need to go in trust in certificate in server side.

jackybek, 2021-08-18:
The certificates are already loaded in /etc/ssl/certs directory, which is the truststore in linux.
What i am observing is a client trying to connect to the server but gets these errors.
I trace the server-side code and found the reason:

     UA_Policy_New_Context() : UA_OpenSSL_loadPrivateKey() fail
     error at UA_SecurityPolicy_Basic256Sha256() : UA_Policy_New_Context() : -2136276992, BadInvalidArgument

It failed during the loading of the private key.
Can I ask if the following procedure to generate openssl privatekey is correct?

     openssl genrsa -out private-key.pem 2048

jackybek, 2021-08-18:
i found an interesting behaviour:
When I choose in UAexpert any one of the following policies and try to connect, the trace log in Open62541 returned the same message:

     Connection 7 | New connection over TCP from 192.168.1.119
     [2021-08-18 12:02:09.681 (UTC+0800)] info/channel       Connection 7 | SecureChannel 9 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
     [2021-08-18 12:02:09.684 (UTC+0800)] info/channel       Connection 7 | SecureChannel 9 | CloseSecureChannel

Basic256Sha256
Aes128_Sha256_RsaOaep
Aes256_Sha256_RsaPss

So this means something went wrong in the open62541 source codes that fail to detect the incoming client request for secured handshake.  Hence the error messages.  It is likely related to the following error during initiatilisation.

     [2021-08-18 11:58:50.638 (UTC+0800)] warn/userland      Could not add SecurityPolicy#Basic256 with error code BadInvalidArgument
     [2021-08-18 11:58:50.638 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
     UA_OpenSSL_LoadPrivateKey() : else segment
     Before exiting UA_OpenSSL_LoadPrivateKey(), result = 0
     UA_Policy_New_Context() : UA_OpenSSL_loadPrivateKey() fail
     error at UA_SecurityPolicy_Basic256Sha256() : UA_Policy_New_Context() : -2136276992, BadInvalidArgument

jackybek, 2021-08-18:
The exact function that fail is 
PEM_read_bio_PrivateKeyPEM_read_bio_PrivateKey(bio, NULL, NULL, NULL) 
in UA_OpenSSL_LoadPrivatekey(const UA_ByteString *privateKey)

jackybek, 2021-08-18:
Resolved.
Issue was the Privatekey was out-of-scope.

liutt1312, 2021-09-29:
> 您需要信任服务器端的证书。

closed 2021-09-29 by liutt1312