open62541/open62541 #4576: PubSub Publish: Sending a NetworkMessage failed : BadEncodingLimitsExceeded

jackybek, 2021-08-03:
## Description
After I added the following statements, i get this error:

#define CA_FILE_PATH_MQTT                      "/etc/ssl/certs/mosq-ca.crt"            // symbolic link => /home/pi/ssl-certs-mosq/mosq-ca.crt
 #define CLIENT_CERT_PATH_MQTT                  "/etc/ssl/certs/brokerserver-11.crt"    // symbolic link => /etc/ssl/certs/brokerserver-11.crt
 #define CLIENT_KEY_PATH_MQTT                   "/etc/ssl/certs/brokerserver-key11.key" // symbolic link => /etc/ssl/certs//brokerserver-key11.key

                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttUseTLS");
                UA_Boolean mqttUseTLS = true;
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttUseTLS, &UA_TYPES[UA_TYPES_BOOLEAN]);
                connectionOptionsIndex++; // 6

                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttCaFilePath");     // or mqttCaPath
                UA_String mqttCaFile = UA_STRING(CA_FILE_PATH_MQTT);
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttCaFile, &UA_TYPES[UA_TYPES_STRING]);
                connectionOptionsIndex++; // 7

                // testing
                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttClientCertPath");
                UA_String mqttClientCertPath = UA_STRING(CLIENT_CERT_PATH_MQTT);
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttClientCertPath, &UA_TYPES[UA_TYPES_STRING]);
                connectionOptionsIndex++; // 8

                // testing
                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttClientKeyPath");
                UA_String mqttClientKeyPath = UA_STRING(CLIENT_KEY_PATH_MQTT);
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttClientKeyPath, &UA_TYPES[UA_TYPES_STRING]);
                connectionOptionsIndex++; // 9

**[2021-08-03 11:35:40.604 (UTC+0800)] error/server       PubSub Publish: Sending a NetworkMessage failed : BadEncodingLimitsExceeded**

I trace the error message to the following function:
 UA_WriterGroup_publishcallback(UA_Server *server, UA_WriterGroup *writerGroup)

Please advise how to resolve this error.

## Background Information / Reproduction Steps

Used CMake options:

-- Cache values
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
[... 131 lines cut ...]
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND
```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): v1.2.2-363-ge1f8fd3b
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system:raspbian linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jackybek, 2021-08-04:
I traced the error and discovered that there is a maximum stack buffer declared for MQTT messages (#define UA_MAX_STACKBUF 128).
In function : sendNetworkMessageJson(), if msgSize > UA_MAX_STACKBUF, it will try to allocate a new buffer using UA_ByteString_allocBuffer(&bug, msgSize).  This is where it fails.

I think there is a logic error in the follow:
============
    /* Allocate the buffer. Allocate on the stack if the buffer is small. */
    UA_ByteString buf;
    size_t msgSize = UA_NetworkMessage_calcSizeJson(&nm, NULL, 0, NULL, 0, true);
    size_t stackSize = 1;
    if(msgSize <= UA_MAX_STACKBUF)
        stackSize = msgSize;
    UA_STACKARRAY(UA_Byte, stackBuf, stackSize);
    buf.data = stackBuf;
    buf.length = msgSize;
    if(msgSize > UA_MAX_STACKBUF) {
        retval = UA_ByteString_allocBuffer(&buf, msgSize);
        if(retval != UA_STATUSCODE_GOOD)
            return retval;
=========
It should be : 
=============
    UA_ByteString buf;
    size_t msgSize = UA_NetworkMessage_calcSizeJson(&nm, NULL, 0, NULL, 0, true);
    size_t stackSize = 1;

    if (msgSize <= UA_MAX_STACKBUF)
    {
        stackSize = msgSize;
        UA_STACKARRAY(UA_Byte, stackBuf, stackSize);
        buf.data = stackBuf;
        buf.length = msgSize;
    }
    else //(msgSize > UA_MAX_STACKBUF)
    {
        retval = UA_ByteString_allocBuffer(&buf, msgSize);
         if(retval != UA_STATUSCODE_GOOD)
            return retval;
    }
============

jpfr, 2021-08-05:
The proposed version wouldn't work.
Because the UA_STACKARRAY variable no longer exists when you quit the if-clause.
So you can't keep the pointer to it.

We can fix this if we can reproduce locally.
Can you show how how the writer is configured?
You can also send a code sample over a private channel if you cannot send it publicly.

jackybek, 2021-08-05:
Julius,
If i trace the proposed code changes, it is the variable (buf) that will hold the data regardless whether it is initialised using UA_STAKKARRAY() or UA_ByteString_alloc().  In both instance, this variable (buf) is accessible after the if-then-else.

Am I correct?

jackybek, 2021-08-07:
i restore the code to its original, made some changes to the my code due to an flag error and now i get this error message:

[2021-08-07 15:26:24.837 (UTC+0800)] info/userland      PubSub MQTT: channel requested.
[2021-08-07 15:26:24.838 (UTC+0800)] error/server       MQTT PubSub: TLS connection requested but open62541 has been built without TLS support

the error message came from connectMqtt(UA_PubSubChannelDataMQTT *) function and it is triggered in the following code segment:

#if defined(UA_ENABLE_MQTT_TLS_OPENSSL) // Extend condition when mbedTLS support is added
    if ((channelData->mqttClientCertPath.length && !channelData->mqttClientKeyPath.length) ||
            (channelData->mqttClientKeyPath.length && !channelData->mqttClientCertPath.length)) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                     "MQTT PubSub: If a client certificate is used, mqttClientCertPath and mqttClientKeyPath must be both specified");
        return UA_STATUSCODE_BADINVALIDARGUMENT;
    }
#else
    if (channelData->mqttUseTLS) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                     **"MQTT PubSub: TLS connection requested but open62541 has been built without TLS support");**
        return UA_STATUSCODE_BADINVALIDARGUMENT;
    }
#endif

How do i enable this flag - **UA_ENABLE_MQTT_TLS_OPENSSL** during the ccmake process?

my current settings in ccmake:
 UA_ENABLE_ENCRYPTION             OPENSSL
 UA_ENABLE_ENCRYPTION_MBEDTLS     OFF
 UA_ENABLE_ENCRYPTION_OPENSSL     ON

Looks like i have to turn on UA_ENABLE)ENCRYPTION_MBEDTLS = ON

jackybek, 2021-08-07:
mosquitto_pub -p 8883 -t AirgardTopic -u myusername -P mypassword --cafile mosq-ca.crt --capath /home/pi/ssl-cert-mosq/ --cert mosq-serv-11.crt --key mosq-serv-11.key --tls-version tlsv1.2 --insecure -m "Message 1"

mosquitto_sub -p 8883 -t AirgardTopic -u myusername -P mypassword --cafile mosq-ca.crt --capath /home/pi/ssl-cert-mosq/ --cert mosq-serv-11.crt --key mosq-serv-11.key --tls-version tlsv1.2 --insecure
**Message 1**

Based on the above parameters, i try to map against the connectionOptions as follows:

        #define CONNECTIONOPTION_NAME_MQTT     "mqttClientId"
        #define USERNAME_OPTION_NAME_MQTT        "mqttUsername"
        #define PASSWORD_OPTION_NAME_MQTT        "mqttPassword"

        int connectionOptionsCount;
        if (MQTT_TLS_Enable == true)
                connectionOptionsCount = 11;
        else
                connectionOptionsCount = 6;

        size_t connectionOptionsIndex = 0;
        UA_KeyValuePair connectionOptions[connectionOptionsCount];
        size_t namespaceIndex;
        printf("1\n");
        UA_Server_getNamespaceByName(uaServer, UA_STRING("virtualskies.com.sg/MKS/"), &namespaceIndex);

        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, CONNECTIONOPTION_NAME_MQTT);
        UA_String mqttClientId = UA_STRING(CLIENT_ID_MQTT);
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttClientId, &UA_TYPES[UA_TYPES_STRING]);
        connectionOptionsIndex++;       // 1

        // login credentials to mqtt broker on 192.168.1.11
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, USERNAME_OPTION_NAME_MQTT);
        UA_String mqttUsername = UA_STRING(USERNAME_MQTT);
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttUsername, &UA_TYPES[UA_TYPES_STRING]);
        connectionOptionsIndex++;       // 2

        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, PASSWORD_OPTION_NAME_MQTT);
        UA_String mqttPassword = UA_STRING(PASSWORD_MQTT);
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttPassword, &UA_TYPES[UA_TYPES_STRING]);
        connectionOptionsIndex++; // 3

        // preallocate sendBufferSize
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "sendBufferSize");
        UA_UInt32 sendBufferSize = 32767;
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &sendBufferSize, &UA_TYPES[UA_TYPES_UINT32]);
        connectionOptionsIndex++; // 4

        // preallocate recvBufferSize
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "recvBufferSize");
        UA_Int32 recvBufferSize = 32767;
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &recvBufferSize, &UA_TYPES[UA_TYPES_UINT32]);
        connectionOptionsIndex++; // 5
        
        printf("2\n");
        if (MQTT_TLS_Enable == true)
        {
                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttUseTLS");
                UA_Boolean mqttUseTLS = true;
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttUseTLS, &UA_TYPES[UA_TYPES_BOOLEAN]);
                connectionOptionsIndex++; // 6

                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttCaFilePath");
                UA_String mqttCaFile = UA_STRING(CA_FILE_PATH_MQTT);
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttCaFile, &UA_TYPES[UA_TYPES_STRING]);
                connectionOptionsIndex++; // 7

                connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "mqttCaPath");
                UA_String mqttCaPath = UA_STRING(CA_PATH_MQTT);
                UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttCaPath, &UA_TYPES[UA_TYPES_STRING]);
                connectionOptionsIndex++; //8
        }
        printf("3\n");
        connectionConfig.connectionProperties = connectionOptions;
        connectionConfig.connectionPropertiesSize = connectionOptionsIndex;
        printf("4\n");

During executing, i get the following errors:

[2021-08-07 17:43:02.779 (UTC+0800)] info/server        PubSub MQTT: Publish
sendNetworkMessageJson(): end of function : Good
[2021-08-07 17:43:08.109 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy channel with openssl is created.
[2021-08-07 17:43:08.165 (UTC+0800)] info/channel       Connection 7 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 and a revised lifetime of 600.00s
[2021-08-07 17:43:08.181 (UTC+0800)] error/server       **PubSub MQTT: yield: error: MQTT_ERROR_SOCKET_ERROR**
[2021-08-07 17:43:14.779 (UTC+0800)] warn/server        PubSub MQTT: sending failed. Invalid state.
sendNetworkMessageJson(): end of function : BadConnectionClosed
[2021-08-07 17:43:14.780 (UTC+0800)] error/server       PubSub Publish: Sending a NetworkMessage failed : BadConnectionClosed
[2021-08-07 17:43:17.499 (UTC+0800)] warn/channel       Connection 7 | SecureChannel 1 | ReadRequest refused without a valid session

Question: What connectionOptions[] parameters did i missed out? or are these correct parameters?

closed 2021-08-29 by jackybek