open62541/open62541 #4580: Compile CNC.NodeSet error

Lo-TW, 2021-08-06:
## Description
I learn to Compile the CNC NodeSet in ubuntu system. 
I use "/open62541/tools/nodeset_compiler/nodeset_compiler.py" to compile "Opc.Ua.CNC.NodeSet.xml", and then  the output are "namespace_cnc.c/.h".
I write a server code, and compile them error: UA_TYPES_CNCPOSITIONDATATYPE’ undeclared.

I don't know how to solve this problem. Could someone give me some suggest？
Thanks！

## Background Information / Reproduction Steps
1. Download open62541
```bash
git clone -b v1.2.2 https://github.com/open62541/open62541.git
```
2. Download submodule(ua-nodeset)
```bash
git submodule update --init
```
3. Edit open62541/CMakeLists.txt
```bash
option(UA_ENABLE_AMALGAMATION "Concatenate the library to a single file open62541.h/.c" ON)

set(UA_NAMESPACE_ZERO "FULL" CACHE STRING "Completeness of the generated namespace zero (minimal/reduced/full)")
```
4. cd open62541/build, and type "cmake .. && make".
 I get the files: libopen62541.a and open62541.h.

5. Compile "Opc.Ua.CNC.NodeSet.xml"
```bash
/usr/bin/python /open62541/tools/nodeset_compiler/nodeset_compiler.py --existing /open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml=/open62541/deps/ua-nodeset/CNC/Opc.Ua.CNC.NodeSet.xml /open62541/build/src_generated/open62541/namespace_cnc
```

I get the files: namespace_cnc.c/.h.

6. Compile my code "server.c" with "namespace_cnc.c". 
Wthen it compile "namespace_cnc.c", its result: error: UA_TYPES_CNCPOSITIONDATATYPE’ undeclared.

my code example：
[myNSCNC.zip](https://github.com/open62541/open62541/files/6943669/myNSCNC.zip)

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tag v1.2.2 / commit ecf5a7037
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 16.04.4 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

GoetzGoerisch, 2021-09-06:
Not instantiating a machine here but my quick sample seams to work with the node set.
https://github.com/umati/Sample-Server/commit/97b26c01d27fd51379b7c23c31292b12ac00bab2

I'm the chairperson for the JWG maintaining the CNC companion specification. Would love to learn your use scenario?

Lo-TW, 2021-09-08:
> Not instantiating a machine here but my quick sample seams to work with the node set.
> [umati/Sample-Server@97b26c0](https://github.com/umati/Sample-Server/commit/97b26c01d27fd51379b7c23c31292b12ac00bab2)
> 
> I'm the chairperson for the JWG maintaining the CNC companion specification. Would love to learn your use scenario?

Hi GoetzGoerisch, 

Thanks for your reply！
Where do I get the Opc.Ua.CNC.NodeSet.bsd in the normal way？
On this https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/CNC , it only provide csv and xml file.

GoetzGoerisch, 2021-09-08:
Normally you generate it from the xml file with a modelling tool.
When the specification was first released a bsd file was not needed and bsd is deprecated anyway.

The nodeset has also some documented errors, if you are starting a new project please get in touch by mail and I happy to help.

NoelGraf, 2021-11-05:
@Lo-TW can the issue be closed or do you still need help?

Lo-TW, 2021-11-26:
If I start the project, we will ask again. Thanks for your help！

closed 2021-11-26 by Lo-TW