open62541/open62541 #4581: Slow addition of many variables

asor, 2021-08-06:
## Slow addition of many variables
I am running an example from the past to check how fast adding node is. Because with version 1.2 I see huge delay when adding many nodes.

```
#include "open62541.h"
#include <signal.h>

static void
addVariable(UA_Server *server, UA_Int32 id) {
    /* Define the attribute of the myInteger variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 my = id;
    UA_Variant_setScalar(&attr.value, &my, &UA_TYPES[UA_TYPES_INT32]);
    char *name = malloc(11);
    sprintf(name, "%07d", id);
    attr.displayName = UA_LOCALIZEDTEXT("en-US", name);
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    UA_NodeId myNodeId = UA_NODEID_NUMERIC(1, id);
    UA_QualifiedName myName = UA_QUALIFIEDNAME(1, name);
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myNodeId, parentNodeId,
                              parentReferenceNodeId, myName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);
    clock_t t;
    double time_taken;
    uint32_t nodes = 1000;
    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    UA_ServerConfig* config = UA_Server_getConfig(server);
    config->verifyRequestTimestamp = UA_RULEHANDLING_ACCEPT;

    t = clock();

    for(int i = 1; i <= nodes; i++)
        addVariable(server, i);
    t = clock() - t;
    time_taken = ((double)t)/CLOCKS_PER_SEC;
    printf("Added %d Variable Nodes for: %f sec.\n",
        nodes, time_taken);
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    return (int)retval;
}
```

Here are some measurements:
Added 1000 Variable Nodes for: 1.504000 sec.
Added 2000 Variable Nodes for: 5.012000 sec.
Added 4000 Variable Nodes for: 18.766000 sec.
Added 8000 Variable Nodes for: 82.781000 sec.

I am running this test on powerful desktop PC.
Do you think the measured values are expected because of complexity of the stack or there is a problem?

EvgKlin, 2021-08-06:
I get: Added 1000 Variable Nodes for: 0.028000 sec.

asor, 2021-08-06:
Which version you are using? I am currently with v1.2.1.

EvgKlin, 2021-08-06:
v1.2.2-253-g4e60f4e5

asor, 2021-08-06:
Thank you EvgKlin! I will try to switch on it.

asor, 2021-08-06:
With the new version works as expected. Thanks again!

closed 2021-08-06 by asor