open62541/open62541 #4582: PubSub over UDP under Windows

busboom, 2021-08-06:
I am trying to get the pubsub examples up and running under Windows 10. The publisher (tuturial_pubsub_publish) works fine and I can see the packets in Wireshark (UDP multicast address 224.0.0.22, port 4840). But the subscriber (tutorial_pubsub_subscribe) never seems to receive anything.

The same issue occurs on native Windows 10, on Ubuntu / WSL1 and on Ubuntu in a VirtualBox under Windows. I have explicitly opened port 4840 for UDP in the Windows firewall configuration.

I have no issues getting the same examples to work on a native Linux system.

Thanks for any advice on this.

johndzilva, 2021-08-09:
How do you test whether your subscriber is receiving data or not?

If you can connect to the Subscriber server with a client tool, you will see that the DateTime variable will be updated frequently (if you are using the tutorial examples)
Or
You can set the UA_LOGLEVEL to 200 (debug) to see the terminal print as "Message received" on successful receive of data.

Ozguryldz, 2021-09-16:
Hello is this issue resolved. I have same issue on ubuntu.
In old versions of open62541 pubsub_subscribe_standalone was succesfully printing the received items for multicast connections.open62541 v1.0
However now it is not receiving any published data.
psc->receive is in different structure with open62541 v1.2