open62541/open62541 #4583: Global Discovery Server

lida-3, 2021-08-06:
Hi,

I'm working on a research project about OPC UA discovery services and I was wondering if GDS is supported by open62541? I was looking into the FEATURES.md but I don't understand the meaning of the new moon and the full moon.
Please let me now if GDS could be implemented with open62541 and if so, does open62541 support all four use cases of certificate management (provisioning, renewal, trust list update and revocation)

Thank you,
Lidija

jpfr, 2021-08-07:
Dear Lidija,

the GDS is not supported at this time.

There is a group working on it. But this takes time to get right.

This needs the GDS itself.
And in addition the necessary hooks in open62541 to enable being the counterpart of a GDS without using private API.

ChHaa, 2021-08-10:
There will be a GDS based on open62541 as well as GDS support, my guess is next year.

Muddasir-Shakil, 2022-06-02:
Dear @ChHaa  and @jpfr,
I wanted to ask about the GDS implementation. When can it be available publicly? Will it also include GDS client? It would be nice to have a GDS client API in Open62 first so that we can also use OPC UA foundation's GDS example for early PoC. 
Thanks.

andreasebner, 2025-02-04:
The GDS Push interface is now part of the latest Master. We are not currently working on the pull mechanism. Updates will be announced in the community call on our mailing list.

closed 2025-02-04 by andreasebner