open62541/open62541 #4585: Check UA_Logger messages through STM32 uart

Maku-chan, 2021-08-08:
## Description
I want to ask that how can I send the messages generated by the UA_Logger to Uart so that I can view them in case of embedded boards.

Thanks.

johndzilva, 2021-08-09:
https://github.com/open62541/open62541/blob/5d7b509d26939065b6e8768c647a53366904b5f9/plugins/ua_log_stdout.c#L51

You can use this function. Inside of this you can work on your UART logic.

jpfr, 2021-08-09:
You can fork the logger plugin and adjust to your use case. It is licensed as CC0.

closed 2021-08-09 by jpfr

Maku-chan, 2021-08-14:
I tried it but it doesnt work. The header file log_stdout.h contains the following

extern UA_EXPORT const UA_Logger UA_Log_Stdout_; /* Logger structure */
extern UA_EXPORT const UA_Logger *UA_Log_Stdout; /* Shorthand pointer */

/* Don't use these definitions. They are only exported as long as the client
 * config is static and required compile-time  */
UA_EXPORT void
UA_Log_Stdout_log(UA_LogLevel level, UA_LogCategory category,
                  const char *msg);
UA_EXPORT void
UA_Log_Stdout_clear(void *logContext);

The logger attribute of the UA_ServerConfig struct is provided the UA_Log_Stdout_ struct which is then used to display all the logging messages so changing the UA_Log_Stdout_log doesnt change it.

UA_Server *
UA_Server_new() {
    UA_ServerConfig config;
    memset(&config, 0, sizeof(UA_ServerConfig));
    /* Set a default logger and NodeStore for the initialization */
    config.logger = UA_Log_Stdout_;
    if(UA_STATUSCODE_GOOD != UA_Nodestore_HashMap(&config.nodestore)) {
        return NULL;
    }

    return UA_Server_newWithConfig(&config);
}