open62541/open62541 #4589: !!! Regression issues after upgrading to v1.2.2-402-g5d7b509d

jackybek, 2021-08-09:
## Description
Previously i reported that  this function : UA_Server_addVariableNode() fails to execute after 10-12 nodes are successfully created.
Then the issue went away. I take it that the programmers discovered a bug and rectify it.

Now this issue came back again.
My application source code did not change.

/*
        // variable
        UA_VariableAttributes vOperatingTimeAttr = UA_VariableAttributes_default;
        vOperatingTimeAttr.description = UA_LOCALIZEDTEXT("en-US", "Operating Time Info");
        vOperatingTimeAttr.displayName = UA_LOCALIZEDTEXT("en-US", "2. Operating Time");
        vOperatingTimeAttr.dataType = UA_TYPES[UA_TYPES_INT16].typeId;
        vOperatingTimeAttr.valueRank = UA_VALUERANK_SCALAR;
        vOperatingTimeAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
        vOperatingTimeAttr.historizing = false;
        UA_Int16 OperatingTime;
        UA_Variant_setScalar(&vOperatingTimeAttr.value, &OperatingTime, &UA_TYPES[UA_TYPES_INT16]);
        retval = UA_Server_addVariableNode(uaServer, UA_NODEID_NUMERIC(namespaceIndex, 10202),
            r2_airgard_status_Id,
            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
            UA_QUALIFIEDNAME(namespaceIndex, "Operating Time"),         //UA_QUALIFIEDNAME(1, "Operating Time"),
            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
            vOperatingTimeAttr, NULL, &outOperatingTime_Id);
        if (retval != UA_STATUSCODE_GOOD)
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "Error creating node : OperatingTime 10202: %s", UA_StatusCode_name(retval));
        else
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "Attribute Variable created : Airgard->StatusType->OperatingTime 10202");               //, r2_airgard_status_Id);
*/

## Background Information / Reproduction Steps
The error I get is segmentation fault.
Are regression tests are performed ?

Program received signal SIGSEGV, Segmentation fault.
0xb6fba42c in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
(gdb) backtrace
#0  0xb6fba42c in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
#1  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2.2-402-g5d7b509d
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: raspbian Linux OPCServer-33 5.10.20-v7l+ #1404 SMP Thu Mar 4 19:44:07 GMT 2021 armv7l GNU/Linux
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

jackybek, 2021-08-09:
-- Cache values
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_READELF:FILEPATH=/usr/bin/readelf
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_RPM:BOOL=OFF
[... 77 lines cut ...]
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=ON
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND
_Python3_EXECUTABLE:FILEPATH=/usr/bin/python3.7

jpfr, 2021-08-09:
It is difficult to help as no debug symbols are in the back trace.

can you use git bisect to find the commit that introduced the issue?

otherwise, if you are on Linux, run the code with valgrind to get better insight to the problem.

jackybek, 2021-08-09:
pi@OPCServer-33:~/OPCUAProject $ ./valgrind --leak-check=yes --show-leak-kinds=all ./myNewServer 192.168.1.33 192.168.1.119 192.168.1.11 8883 pub
==11725== Memcheck, a memory error detector
==11725== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==11725== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==11725== Command: ./myNewServer 192.168.1.33 192.168.1.119 192.168.1.11 8883 pub
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4866210: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x48662DC: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x73CA638: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4863B6C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4864F44: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Use of uninitialised value of size 4
==11725==    at 0x4864F54: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x48662E4: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x73CA694: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4864F1C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4864F38: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x486505C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4865074: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x486508C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x48650A0: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x48650BC: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x48650C8: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x73CD8AC: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x73CD8CC: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
==11725== Invalid read of size 8
==11725==    at 0x4866214: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==  Address 0x766ec28 is 0 bytes after a block of size 24 alloc'd
==11725==    at 0x48476D4: malloc (vg_replace_malloc.c:307)
==11725==    by 0x73CAA67: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
==11725== Invalid read of size 8
==11725==    at 0x4865094: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==  Address 0x767096c is 20 bytes inside a block of size 26 alloc'd
==11725==    at 0x48476D4: malloc (vg_replace_malloc.c:307)
[... 57 lines cut ...]
==11725==    at 0x4865148: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Conditional jump or move depends on uninitialised value(s)
==11725==    at 0x4865158: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==
==11725== Invalid read of size 8
==11725==    at 0x4865004: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so)
==11725==  Address 0x768c17c is 20 bytes inside a block of size 26 alloc'd
==11725==    at 0x48476D4: malloc (vg_replace_malloc.c:307)
==11725==    by 0x73CAA67: ??? (in /usr/lib/arm-linux-gnueabihf/libtasn1.so.6.5.5)
==11725==
disInstr(arm): unhandled instruction: 0xEC510F1E
                 cond=14(0xE) 27:20=197(0xC5) 4:4=1 3:0=14(0xE)
==11725== valgrind: Unrecognised instruction at address 0x4ce8728.
==11725==    at 0x4CE8728: ??? (in /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1)
==11725== Your program just tried to execute an instruction that Valgrind
==11725== did not recognise.  There are two possible reasons for this.
==11725== 1. Your program has a bug and erroneously jumped to a non-code
==11725==    location.  If you are running Memcheck and you just saw a
==11725==    warning about a bad jump, it's probably your program's fault.
==11725== 2. The instruction is legitimate but Valgrind doesn't handle it,
==11725==    i.e. it's Valgrind's fault.  If you think this is the case or
==11725==    you are not sure, please let us know and we'll try to fix it.
==11725== Either way, Valgrind will now raise a SIGILL signal which will
==11725== probably kill your program.

jackybek, 2021-08-09:
==11725== 576 bytes in 2 blocks are still reachable in loss record 25,195 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x25DCB: UA_Array_copy (open62541.c:15398)
==11725==    by 0x23D1B: Variant_copy (open62541.c:14649)
==11725==    by 0x258CB: UA_copy (open62541.c:15258)
==11725==    by 0x16547: UA_Variant_copy (open62541.h:23339)
==11725==    by 0x37A9F: copyCommonVariableAttributes (open62541.c:34636)
==11725==    by 0x37B47: copyVariableNodeAttributes (open62541.c:34649)
==11725==    by 0x37E03: UA_Node_setAttributes (open62541.c:34720)
==11725==    by 0x71DC7: AddNode_raw (open62541.c:55357)
==11725==    by 0x72313: Operation_addNode_begin (open62541.c:55468)
==11725==    by 0x73A97: UA_Server_addNode_begin (open62541.c:55953)
==11725==    by 0xA1203: function_namespace0_generated_198_begin (open62541.c:68618)
==11725==
==11725== 592 bytes in 2 blocks are still reachable in loss record 25,196 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x416ABB: createEntry (open62541.c:211109)
==11725==    by 0x416D57: UA_NodeMap_newNode (open62541.c:211170)
==11725==    by 0x71C73: AddNode_raw (open62541.c:55338)
==11725==    by 0x72313: Operation_addNode_begin (open62541.c:55468)
==11725==    by 0x736EB: Operation_addNode (open62541.c:55852)
==11725==    by 0x73907: addNode (open62541.c:55908)
==11725==    by 0x739C3: __UA_Server_addNode (open62541.c:55928)
==11725==    by 0x18C03: UA_Server_addVariableTypeNode (open62541.h:38393)
==11725==    by 0x3B32B: UA_Server_createNS0_base (open62541.c:35958)
==11725==    by 0x3CCA7: UA_Server_initNS0 (open62541.c:36561)
==11725==    by 0x3949F: UA_Server_init (open62541.c:35335)
==11725==
==11725== 592 bytes in 2 blocks are still reachable in loss record 25,197 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x416ABB: createEntry (open62541.c:211109)
==11725==    by 0x416D57: UA_NodeMap_newNode (open62541.c:211170)
==11725==    by 0x71C73: AddNode_raw (open62541.c:55338)
==11725==    by 0x72313: Operation_addNode_begin (open62541.c:55468)
==11725==    by 0x736EB: Operation_addNode (open62541.c:55852)
==11725==    by 0x73907: addNode (open62541.c:55908)
==11725==    by 0x739C3: __UA_Server_addNode (open62541.c:55928)
==11725==    by 0x18C03: UA_Server_addVariableTypeNode (open62541.h:38393)
==11725==    by 0x3B497: UA_Server_createNS0_base (open62541.c:35967)
==11725==    by 0x3CCA7: UA_Server_initNS0 (open62541.c:36561)
==11725==    by 0x3949F: UA_Server_init (open62541.c:35335)
==11725==
==11725== 600 bytes in 2 blocks are still reachable in loss record 25,198 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x25DCB: UA_Array_copy (open62541.c:15398)
==11725==    by 0x23D1B: Variant_copy (open62541.c:14649)
==11725==    by 0x258CB: UA_copy (open62541.c:15258)
==11725==    by 0x16547: UA_Variant_copy (open62541.h:23339)
==11725==    by 0x37A9F: copyCommonVariableAttributes (open62541.c:34636)
==11725==    by 0x37B47: copyVariableNodeAttributes (open62541.c:34649)
==11725==    by 0x37E03: UA_Node_setAttributes (open62541.c:34720)
==11725==    by 0x71DC7: AddNode_raw (open62541.c:55357)
==11725==    by 0x72313: Operation_addNode_begin (open62541.c:55468)
==11725==    by 0x73A97: UA_Server_addNode_begin (open62541.c:55953)
==11725==    by 0x10B75B: function_namespace0_generated_882_begin (open62541.c:88999)
==11725==
==11725== 600 bytes in 2 blocks are still reachable in loss record 25,199 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x25DCB: UA_Array_copy (open62541.c:15398)
==11725==    by 0x23D1B: Variant_copy (open62541.c:14649)
==11725==    by 0x258CB: UA_copy (open62541.c:15258)
==11725==    by 0x16547: UA_Variant_copy (open62541.h:23339)
==11725==    by 0x37A9F: copyCommonVariableAttributes (open62541.c:34636)
==11725==    by 0x37B47: copyVariableNodeAttributes (open62541.c:34649)
==11725==    by 0x37E03: UA_Node_setAttributes (open62541.c:34720)
==11725==    by 0x71DC7: AddNode_raw (open62541.c:55357)
==11725==    by 0x72313: Operation_addNode_begin (open62541.c:55468)
==11725==    by 0x73A97: UA_Server_addNode_begin (open62541.c:55953)
==11725==    by 0x10E8B7: function_namespace0_generated_897_begin (open62541.c:89679)
==11725==
[... 1954 lines cut ...]
==11725== 901,241 bytes in 603 blocks are still reachable in loss record 25,350 of 25,350
==11725==    at 0x484A2F4: calloc (vg_replace_malloc.c:760)
==11725==    by 0x25DCB: UA_Array_copy (open62541.c:15398)
==11725==    by 0x2267F: String_copy (open62541.c:14193)
==11725==    by 0x258CB: UA_copy (open62541.c:15258)
==11725==    by 0x25E77: UA_Array_copy (open62541.c:15411)
==11725==    by 0x23D1B: Variant_copy (open62541.c:14649)
==11725==    by 0x258CB: UA_copy (open62541.c:15258)
==11725==    by 0x16547: UA_Variant_copy (open62541.h:23339)
==11725==    by 0x37A9F: copyCommonVariableAttributes (open62541.c:34636)
==11725==    by 0x37B47: copyVariableNodeAttributes (open62541.c:34649)
==11725==    by 0x37E03: UA_Node_setAttributes (open62541.c:34720)
==11725==    by 0x71DC7: AddNode_raw (open62541.c:55357)
==11725==
==11725== LEAK SUMMARY:
==11725==    definitely lost: 275 bytes in 2 blocks
==11725==    indirectly lost: 0 bytes in 0 blocks
==11725==      possibly lost: 770,051 bytes in 11,326 blocks
==11725==    still reachable: 4,665,638 bytes in 62,887 blocks
==11725==         suppressed: 0 bytes in 0 blocks
==11725==
==11725== Use --track-origins=yes to see where uninitialised values come from
==11725== For lists of detected and suppressed errors, rerun with: -s
==11725== ERROR SUMMARY: 85798 errors from 5429 contexts (suppressed: 0 from 0)
Segmentation fault

jackybek, 2021-08-22:
Issued is resolved by removing the following during UA_Server_addVariableNode():

        //char SoftwareVersion[MAX_STRING_SIZE] = "open62541";
	//if (!UA_Variant_isEmpty(&vSoftwareVersionAttr.value))
	        //UA_Variant_setScalar(&vSoftwareVersionAttr.value, &SoftwareVersion, &UA_TYPES[UA_TYPES_STRING]);
        //else UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
        //        "Error UA_Variant_setScalar() : SoftwareVersion 10001");

closed 2021-08-22 by jackybek