open62541/open62541 #4590: UA_VariableNode  v:

embbo, 2021-08-09:
I am trying to use this dataType  UA_VariableNode  "containing all the attributes" and want to see the on UAexpert  but what  is the UA_TYPES[UA_TYPES_?] of it 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-08-09:
I think this datatype is filtered out from the generation. UA_VariableNode is an internal data structure and not generated

jpfr, 2021-08-09:
What are yo trying to achieve?

embbo, 2021-08-09:
@jpfr I am trying to parse XML at client side and want to send all the attributes to the embedded server as structure and save in EEPROM .. The only data type , which has all the attributes is this one. "UA_VariableNode"