open62541/open62541 #4591: Question: Can DataSource variable nodes have no value when not of BaseDataType?

JensMertelmeyer, 2021-08-09:
## Question
I add my variable nodes with `UA_Server_addDataSourceVariableNode(..)`, i love data sources 😊
However, in some scenarios, I have no data to return.

Consider this trimmed down snippet:

```cpp
#include <open62541/server.h>
#include <open62541/server_config_default.h>

void addVariable(UA_Server* server) {
	UA_DataSource dataSource;
	dataSource.write = nullptr;
	dataSource.read = handleCallbackRead;
	
	UA_VariableAttributes attributes = UA_VariableAttributes_default;
	attributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_STRING);
	(...)
	
	UA_Server_addDataSourceVariableNode(
		(...)
		attributes, dataSource,
		(...)
	);
}	
	
int main(int argc, char **argv) {
	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addVariable(server);
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}
```

I thought I could just set the `handleCallbackRead` to something like this:
```cpp
UA_StatusCode handleCallbackRead(
	UA_Server *server, const UA_NodeId *sessionId,
	void *sessionContext, const UA_NodeId *nodeId,
	void *nodeContext, UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
) {
	dataValue->hasValue = false;
	return UA_STATUSCODE_GOOD;
}
```

but  `UA_Server_addDataSourceVariableNode` will then produce the following log message: 

> info/session | AddNode (ns=1;s=myNodeName): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.

It will then try to write the value and fail, because the node is not writeable.
I thought maybe I could return something other than `UA_STATUSCODE_GOOD`, for example `UA_STATUSCODE_GOODNODATA`. 

I was going through the implementation source, but it seems like `UA_STATUSCODE_GOOD` is the only expected return value.

https://github.com/open62541/open62541/blob/68d8c3e419782a53a8210a3a98c705889260eef3/src/server/ua_services_nodemanagement.c#L2271-L2273

## Possibly related
Issue #914 already had a very similar question, but there were no data sources involved.
@jpfr said:
> Hmm. The standard defines a second mechanism for (dis-)allowing null values.
> That gives us a bit more leeway for our implementation.
> (...)
> We could allow null values for every data type.

See https://github.com/open62541/open62541/issues/914#issuecomment-267184455

## Complete Source & Log
<details>
  <summary>Here is the complete code (70 lines)</summary>

```cpp
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/log_stdout.h>
void addVariable(UA_Server* server);

UA_StatusCode handleCallbackRead(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
);

int main(int argc, char **argv) {
	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addVariable(server);
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}
	
void addVariable(UA_Server* server) {
	const auto TYPEDEF_VAR          = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
	
	const auto REFTYPE_HASCOMPONENT = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
	const UA_NodeId OBJECTS_FOLDER  = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_DataSource dataSource;
	dataSource.write = nullptr;
	dataSource.read = handleCallbackRead;
	
	UA_VariableAttributes attributes = UA_VariableAttributes_default;
	attributes.displayName = UA_LOCALIZEDTEXT_ALLOC("en", "My Text");
	attributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_STRING);
	attributes.valueRank = UA_VALUERANK_SCALAR;
	attributes.accessLevel = UA_ACCESSLEVELMASK_READ;
	
	UA_NodeId newNodeId = UA_NODEID_STRING_ALLOC(1, "MyText");
	auto ret = UA_Server_addDataSourceVariableNode(
		server, newNodeId, 
		OBJECTS_FOLDER, REFTYPE_HASCOMPONENT,
		UA_QUALIFIEDNAME_ALLOC(1, "MyText"),
		TYPEDEF_VAR,
		attributes, dataSource,
		nullptr /*nodeContext*/, nullptr /*out newNodeId*/
	);
	UA_LOG_INFO(
		UA_Log_Stdout, 
		UA_LOGCATEGORY_SERVER, 
		"addDataSourceVariableNode(..): 0x%x (%s)", ret, UA_StatusCode_name(ret)
	); 
}
	
UA_StatusCode handleCallbackRead(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
) {
	dataValue->hasValue = false;
	return UA_STATUSCODE_GOOD;
}
```
</details>

<details>
<summary>Here is the log output</summary>

```
info/session | AddNode (ns=1;s=MyText): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
info/session | WriteRequest returned status code BadWriteNotSupported
info/session | AddNode (ns=1;s=MyText): Could not create a default value with StatusCode BadWriteNotSupported      
info/session | Only Variables with data type BaseDataType can contain an empty value
info/session | WriteRequest returned status code BadWriteNotSupported
info/session | AddNode (ns=1;s=MyText): The value is incompatible with the variable definition
info/session | AddNode (ns=1;s=MyText): Type-checking failed with error code BadWriteNotSupported
```

</details>

## Bonus Chatter
- I thought that `BaseDataType` was just fine. I tried several OPC UA clients, they all read the data just fine
- Now I have a client with some "Siemens TIA" software, that apparently cannot cope with that. It seems that it needs the variables to be of type `UA_String`, `UA_Float` and so on.
- Because of that, I think I will have to change all variable types.
- Unfortunately, I sometimes return null values. And this only seems possible for `BaseDataType`
- I somebody knows a way to keep "nullable" values and still be compatible with that "Siemens TIA" software, I'm all ears 🙂

## Checklist
 - [x] open62541 Version (release number or git tag): **1.2**
 - [x] Other OPC UA SDKs used (client or server): **None**
 - [x] Operating system: **Windows x64**
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: **yes**
 - [x] Wireshark network dump attached: **None needed**
 - [x] Self-contained code example attached: **Yes**
 - [x] Critical issue: **No**

Thanks for reading.

ccvca, 2021-08-10:
The returned value is not the StatusCode of the DataValue. This should be possible using the appropiate members:
https://github.com/open62541/open62541/blob/cbc7d0dbfe7db137271e5c842a8b465c59045516/include/open62541/types.h#L881-L900

JensMertelmeyer, 2021-08-10:
Thank you for your reply. I've had no success with setting the status of the value.

It seems that returning a **null value with data source nodes is not possible**, while it is perfectly **possible with regular nodes**.

It seems that when first adding the node, a value _must_ exist. I will see if I can make my data source handlers return a default value, and when all nodes have been added, then return null values, when appropriate.

jpfr, 2021-08-10:
Yes, only BaseDataType can be null.
You have to supply a value for all other types.

What you see is a convenience function to set default values if none are supplied.

We can add a configuration option for this. So that setting the default values may fail.

JensMertelmeyer, 2021-08-10:
Thank you for your explanation.
I'm afraid I still don't quite understand. The way I see it is:

1. When creating a node that is not `BaseDataType`, a value **must** be supplied
2. Right after adding the node, it is legal to change the value to an empty variant (applies to both DataSource and regular node)

Is that correct? You make it sound like it is not legal.

I tried to keep it as short as possible, here is a snippet that demonstrates it both for a DataSource and a regular variable node:

<details>
<summary>Code (145 lines)</summary>

```cpp
#include <string>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/log_stdout.h>
	
#pragma region forward declarations
void addDataSourceVariable(UA_Server *server, const std::string nodeName);
void addRegularVariable(UA_Server *server, const std::string nodeName);
UA_StatusCode datasource_read_emptystr(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
);
UA_StatusCode datasource_read_nullvalue(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
);
#pragma endregion
	
#pragma region consts
const auto TYPEDEF_VAR          = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
const auto REFTYPE_HASCOMPONENT = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
const UA_NodeId OBJECTS_FOLDER  = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
#pragma endregion
	
int main(int argc, char **argv) {
	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addDataSourceVariable(server, "myDataSourceNode");
	addRegularVariable(server, "myRegularNode");
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}
	
void addDataSourceVariable(UA_Server *server, const std::string nodeName) {
	UA_VariableAttributes attributes = UA_VariableAttributes_default;
	attributes.displayName = UA_LOCALIZEDTEXT_ALLOC("en", nodeName.c_str());
	attributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_STRING);
	attributes.valueRank = UA_VALUERANK_SCALAR;
	attributes.accessLevel = UA_ACCESSLEVELMASK_READ;
	
	UA_NodeId newNodeId = UA_NODEID_STRING_ALLOC(1, nodeName.c_str());
	
	// First, create the variable with a datasource that will return
	// an empty string
	UA_DataSource dataSource;
	dataSource.write = nullptr;
	dataSource.read = datasource_read_emptystr;	
	
	UA_Server_addDataSourceVariableNode(
		server, newNodeId, 
		OBJECTS_FOLDER, REFTYPE_HASCOMPONENT,
		UA_QUALIFIEDNAME_ALLOC(1, nodeName.c_str()),
		TYPEDEF_VAR,
		attributes, dataSource,
		nullptr /*nodeContext*/, nullptr /*out newNodeId*/
	);
	
	// Then, set the datasource handler back to one that returns null
	dataSource.read = datasource_read_nullvalue;
	UA_Server_setVariableNode_dataSource(server, newNodeId, dataSource);
}

void addRegularVariable(UA_Server *server, const std::string nodeName) {
	UA_VariableAttributes attributes = UA_VariableAttributes_default;
	attributes.displayName = UA_LOCALIZEDTEXT_ALLOC("en", nodeName.c_str());
	attributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_STRING);
	attributes.valueRank = UA_VALUERANK_SCALAR;
	attributes.accessLevel = UA_ACCESSLEVELMASK_READ;

	UA_NodeId newNodeId = UA_NODEID_STRING_ALLOC(1, nodeName.c_str());
	
	// Supply an empty string as a starting value
	UA_String str;
	UA_String_init(&str); 
	UA_Variant_setScalarCopy(&attributes.value, &str, &UA_TYPES[UA_TYPES_STRING]);
	
	UA_Server_addVariableNode(
		server, newNodeId,
		OBJECTS_FOLDER, REFTYPE_HASCOMPONENT,
		UA_QUALIFIEDNAME_ALLOC(1, nodeName.c_str()),
		TYPEDEF_VAR,
		attributes,
		nullptr /*nodeContext*/, nullptr /*outNewNodeId*/
	);
	
	// Then, set the value back to null
	UA_Variant v;
	UA_Variant_clear(&v);
	UA_Server_writeValue(server, newNodeId, v);
}
	
UA_StatusCode datasource_read_emptystr(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
) {
	dataValue->hasValue = true;
	
	UA_String str;
	UA_String_init(&str); 
	UA_Variant_setScalarCopy(&dataValue->value, &str, &UA_TYPES[UA_TYPES_STRING]);
	
	dataValue->serverTimestamp = UA_DateTime_now();
	dataValue->hasServerTimestamp = true;
	dataValue->hasSourceTimestamp = false;  
	
	return UA_STATUSCODE_GOOD;
}

UA_StatusCode datasource_read_nullvalue(
	UA_Server *server,
	const UA_NodeId *sessionId,
	void *sessionContext,
	const UA_NodeId *nodeId,
	void *nodeContext,
	UA_Boolean sourceTimeStamp,
	const UA_NumericRange *range,
	UA_DataValue *dataValue
) {
	dataValue->hasValue = false;
	
	// not sure if this is appropriate
	dataValue->hasStatus = true;
	dataValue->status = UA_STATUSCODE_GOODNODATA;
	
	dataValue->serverTimestamp = UA_DateTime_now();
	dataValue->hasServerTimestamp = true;
	dataValue->hasSourceTimestamp = false;  
	
	return UA_STATUSCODE_GOOD;
}
```
</details>

UaExpert, for example, will correctly display that it's of type "String" and has a null value.
<details><summary>Screenshot</summary>
[image]
</details>

JensMertelmeyer, 2021-09-03:
I have received confirmation that my client is now able to read the values in his Siemens-thing.

I am still a bit puzzled why I **must** supply a value when adding the variable, but it appears to be perfectly fine to set it back to `NULL` directly afterwards. To me, it looks unnecessary.

Maybe @jpfr can comment once more on my approach [from above](https://github.com/open62541/open62541/issues/4591#issuecomment-895858657)? I still haven't quite understood if what I do is frowned upon, downright illegal or perfectly fine.

closed 2021-09-03 by JensMertelmeyer