open62541/open62541 #4598: Is UA_Client thread-safe?

tboeckel, 2021-08-16:
## Description
Are all the UA_Client_xxx() functions thread-safe? I need to call several client functions from different threads. Currently I protect these calls (including the central UA_Client_run_iterate() call) with a mutex. Is this necessary at all or can the mutex locking be omitted?

## Background Information / Reproduction Steps
My application implements both a server and a client. The client connects to another server. Each method call on my application's server must be forwarded to the corresponding method of the second server. This chained call usually takes approx 100ms to finish, which is absolutely ok from my point of view. But sometimes it takes up to 4 or 5 seconds, and in very rare situation the calling application runs into a timeout (usually after 10 seconds).

After some investigation I found out that the delay was caused by waiting for the mutex protecting the UA_Client_run_iterate() call with an iteration timeout of just 10ms. I expect that call to return after 10ms at most and hence the mutex should not be locked much longer than 10ms. But since the mutex protected call of UA_Client_call() is sometimes delayed much longer (up to 10 seconds) the only explanation is that the time is spent in UA_Client_run_iterate() as this is the only other call protected by the same mutex.

I tried to run the client code without the mutex protection, and so far it works without any bad impact. But on the other hand some months back I ran into REALLY strange problems and crashes when calling UA_Server_xxx() functions from different threads. This all was cured either by protecting certain call with a mutex or by using UA_MULTITHREADING=100. But according to the source code all this internal multithreading support is about UA_Server, not about UA_Client.

So my questions is: is UA_Client thread-safe regarding function calls from different threads by nature? Or must thread-safety be enforced artificially (i.e. by a custom mutex)?

Used CMake options:
UA_ENABLE_AMALGAMATION=ON
UA_MULTITHREADING=100

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.2.2
 - [X] Other OPC UA SDKs used (client or server): UaExpert
 - [X] Operating system: Ubuntu Linux, Yocto Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-08-16:
Functions that are safe to be called from parallel threads are marked UA_THREADSAFE.
This is the case for most functions for the server. And none for the client.
So no, we don't have thread-safety for the client today. (But it will come eventually.)

You can workaround by preticting the client with an external lock.

closed 2021-08-16 by jpfr

tboeckel, 2021-08-16:
Thanks for the clarification!
But is there a way to tell why UA_Client_run_iterate() takes so much longer to return than allowed (10ms vs 10s)? I mean, why do we have a restricted maximum running time if it is not always obeyed?

Basically I have these two code parts running in different threads:
`// the client's main loop`
`clientMutex.lock();`
`UA_Client_run_iterate(client, 10);`
clientMutex.unlock();`

`// the method calling thread`
`clientMutex.lock();`
`UA_Client_call_async(client, ...);`
`clientMutex.unlock();`

I also tried to obtain the mutex with a timeout of 100ms and print the owner in case of failure. Guess what: the mutex was always locked due to the UA_Client_run_iterate() call. But that one is limited to 10ms runtime. So why does the call take longer than allowed?

tboeckel, 2021-08-17:
Forget my last comment. As it seems the problem is located somewhere outside open62541. The UA_Client_run_iterate() call  with 10ms timeout usually returns within 9ms, very seldom it takes one or two milliseconds more. The "hang" occurs somewhere in the mutex locking, which is beyond open62541's scope.

Nevertheless, many thanks for your help!