open62541/open62541 #4599: Missing functions in win32 (x64) build of 1.2

skolodzieski, 2021-08-16:
## Description

## Background Information / Reproduction Steps
Build a win32 (x64) version of the open62541 library with mbedtls encryption. Used the same recipe I used on linux x86_64 and linux aarch64. Some functions are not defined in the static library:

```
open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ServerNetworkLayerTCP referenced in function UA_ServerConfig_addNetworkLayerTCP
open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ClientConnectionTCP_init referenced in function UA_ClientConfig_setDefault
open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ClientConnectionTCP_poll referenced in function UA_ClientConfig_setDefault
mbedcrypto.lib(entropy_poll.obj) : error LNK2019: unresolved external symbol __imp_CryptAcquireContextA referenced in function mbedtls_platform_entropy_poll
mbedcrypto.lib(entropy_poll.obj) : error LNK2019: unresolved external symbol __imp_CryptReleaseContext referenced in function mbedtls_platform_entropy_poll
mbedcrypto.lib(entropy_poll.obj) : error LNK2019: unresolved external symbol __imp_CryptGenRandom referenced in function mbedtls_platform_entropy_poll
```

I will investigate the unresolves mbedtls functions, but ``` UA_XXX functions``` are the ones of immediate concern. 

Used CMake options:

```bash
cmake -G "Visual Studio 12 Win64" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/c/home/sckolo/work/libs/open62541/build-win64-tls/install -DUA_ENABLE_ENCRYPTION_MBEDTLS=ON -DMBEDTLS_INCLUDE_DIRS=c:/home/sckolo/work/libs/mbedtls/build-win64/install/include  -DMBEDTLS_LIBRARY=c:/home/sckolo/work/libs/mbedtls/build-win64/install/lib/mbedtls.lib  -DMBEDX509_LIBRARY=c:/home/sckolo/work/libs/mbedtls/build-win64/install/lib/mbedx509.lib -DMBEDCRYPTO_LIBRARY=c:/home/sckolo/work/libs/mbedtls/build-win64/install/lib/mbedcrypto.lib -DUA_LOGLEVEL=700 -DUA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

```bash
[build-win64-tls-1.2]$ git branch --list
* (HEAD detached at v1.2)
```

skolodzieski, 2021-08-16:
Since this is a build / link issue, there are no logs, network dump or code examples.

andreasebner, 2025-07-30:
This is tested in CI now, so it should be resolved.

closed 2025-07-30 by andreasebner