open62541/open62541 #4600: Add Engineering Units

vvpnet, 2021-08-17:
## Description
Hello everyone. Tell me how to add engineering units to my node?
Му Node create:

   ```
 bool isOk = true;

    UA_VariableAttributes valueAttr = UA_VariableAttributes_default;
    UA_Variant_init(&valueAttr.value);

    UA_Float valueVal = 0;
    UA_Variant_setScalar(&valueAttr.value, &valueVal, &UA_TYPES[UA_TYPES_FLOAT]);

    std::string description = "Value Params";
    valueAttr.description = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>(description.c_str()));
    valueAttr.displayName = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>("Value"));

    if(UA_Server_addVariableNode(server, UA_NODEID_NULL,
                                 getNodeId(),
                                 UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                                 UA_QUALIFIEDNAME(nsIndexBrowser, const_cast<char*>("Value")),
                                 UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                                 valueAttr, nullptr, &value_) != UA_STATUSCODE_GOOD)
    {
        return isOk;
    }

```

I tried to do this:

```
    UA_EUInformation eu;
    eu.namespaceUri = UA_STRING("somecompany.com");
    eu.unitId = -1;
    eu.displayName = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>(typeSigne_->getEUid().c_str()));
    eu.description = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>(typeSigne_->getEUid().c_str()));

    UA_StatusCode status = UA_Server_writeObjectProperty_scalar(server,
                                                                value_,
                                                                UA_QUALIFIEDNAME(nsIndexBrowser, "EngineeringUnits"),
                                                                &eu,
                                                                &UA_TYPES[UA_TYPES_EUINFORMATION]);

    if(status != UA_STATUSCODE_GOOD)
    {
        return isOk;
    }

```
But I got the error code UA_STATUSCODE_BADNOMATCH

I tried to do this:

```
    UA_EUInformation eu;
    eu.namespaceUri = UA_STRING("somecompany.com");
    eu.unitId = -1;
    eu.displayName = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>(typeSigne_->getEUid().c_str()));
    eu.description = UA_LOCALIZEDTEXT(const_cast<char*>(lt), const_cast<char*>(typeSigne_->getEUid().c_str()));
    UA_VariableAttributes enAttr = UA_VariableAttributes_default;
    UA_VariableAttributes_init(&enAttr);
    enAttr.dataType = UA_TYPES[UA_TYPES_EUINFORMATION].typeId;
    enAttr.userWriteMask = UA_ACCESSLEVELMASK_READ;
    UA_Variant_setScalarCopy(&enAttr.value, &eu, &UA_TYPES[UA_TYPES_EUINFORMATION]);

    UA_StatusCode status = UA_Server_addVariableNode(server, UA_NODEID_NULL,
                                                     value_,
                                                     UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
                                                     UA_QUALIFIEDNAME(nsIndexBrowser, "EngineeringUnits"),
                                                     UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
                                                     enAttr,
                                                     nullptr, &enUnitNodeUnits_);
    if(status != UA_STATUSCODE_GOOD)
    {
        return isOk;
    }
```

But I got the error code UA_STATUSCODE_BADTYPEMISMATCH

I don't understand why I can't do it?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [1 ] open62541 Version (release number or git tag): 1.1.2-216
 - [ ] Other OPC UA SDKs used (client or server): 
 - [1 ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

vvpnet, 2021-08-24:
I also tried these examples [https://groups.google.com/g/open62541/c/90dpSO2Wu7E/m/mNUknnRSCAAJ](url) . They don't work.

Muddasir-Shakil, 2021-08-24:
Hi, 
In the first case you have to chose UA_NODEID_NUMERIC(0, UA_NS0ID_ANALOGITEMTYPE)[1], instead of  UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE) because baseDataVariable does not have EUInfo property[2]. 
 Ref: [1] https://reference.opcfoundation.org/v104/Core/VariableTypes/BaseAnalogType/ 
        [2] https://reference.opcfoundation.org/v104/Core/VariableTypes/BaseDataVariableType/

vvpnet, 2021-08-30:
> Hi,
> In the first case you have to chose UA_NODEID_NUMERIC(0, UA_NS0ID_ANALOGITEMTYPE)[1], instead of UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE) because baseDataVariable does not have EUInfo [...]

Hi Muddasir-Shakil. Thank you very much for the answer and  sorry for my bad English.
I have already tried this and got an error "BadTypeMismatch". Аt the time of adding the node "valueAttr".

This is the parent of the node "valueAttr":
```
    UA_NodeId parentNodeId = getParentNodeId();
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(ns::Index, UA_NS0ID_ORGANIZES);

    std::string strBrowseName = getNameTranslate();
    UA_QualifiedName browseName = UA_QUALIFIEDNAME(ns::qIndexBrowser, const_cast<char*>(strBrowseName.c_str()));

    UA_NodeId typeDefinition = UA_NODEID_NUMERIC(ns::Index, UA_NS0ID_BASEOBJECTTYPE);

    UA_StatusCode retval = UA_Server_addObjectNode(server,
                                                   UA_NODEID_NULL,
                                                   parentNodeId,
                                                   parentReferenceNodeId,
                                                   browseName,
                                                   typeDefinition,
                                                   oAttr, this,
                                                   &nodeId_);

```

dns13, 2022-06-29:
I've got it somehow working with combining multiple approaches I've found online.

I'm using `UA_NS0ID_ANALOGITEMTYPE` for the TypeDefinition, adding the EngineeringUnits as a node manually and using `UA_Server_addReference()` to link it to the variable node.

But I'm not sure if this is on spec as it works different as EURange where just the `UA_Server_writeObjectProperty_scalar()` is doing perfectly fine.

Here is the full code I'm using:
```C
template <typename T>
auto addVariableNode(UA_Server *server, const UA_NodeId *parent_node_id, T *value, const std::string &display_name,
					const std::string &description, const std::string &qualified_name, bool allow_write,
					UA_EUInformation *eu) -> UA_NodeId {
auto *type = getUADataType<T>();
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.dataType = type->typeId;
attr.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>(description.c_str()));
attr.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>(display_name.c_str()));
attr.accessLevel = UA_ACCESSLEVELMASK_READ | (allow_write ? UA_ACCESSLEVELMASK_WRITE : 0u);

UA_QualifiedName new_qualified_name = UA_QUALIFIEDNAME(1, const_cast<char*>(qualified_name.c_str()));

UA_NodeId parent_node_id_temp;

if (parent_node_id == nullptr) {
	parent_node_id_temp = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
} else {
	parent_node_id_temp = *parent_node_id;
}

UA_NodeId parent_reference_node_id = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
UA_NodeId variable_type_node_id = UA_NODEID_NUMERIC(0, UA_NS0ID_ANALOGITEMTYPE);

UA_DataSource data_source;
data_source.read = readData<T>;
data_source.write = writeData<T>;

UA_NodeId new_node_id;
auto retval = UA_Server_addDataSourceVariableNode(server, UA_NODEID_NULL, parent_node_id_temp, parent_reference_node_id,
									new_qualified_name, variable_type_node_id, attr, data_source, value,
									&new_node_id);

if (retval != UA_STATUSCODE_GOOD) {
	UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%s: %s", new_qualified_name.name.data,
					UA_StatusCode_name(retval));
}

new_qualified_name = UA_QUALIFIEDNAME(0, const_cast<char*>("EURange"));

UA_Range minmax = {.low = 0.0, .high = 100.0};

retval = UA_Server_writeObjectProperty_scalar(server, new_node_id, new_qualified_name, &minmax,
													&UA_TYPES[UA_TYPES_RANGE]);

if (retval != UA_STATUSCODE_GOOD) {
	UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%s: %s", new_qualified_name.name.data,
					UA_StatusCode_name(retval));
}

new_qualified_name = UA_QUALIFIEDNAME(0, const_cast<char*>("EngineeringUnits"));

UA_NodeId temp_units;

UA_VariableAttributes enAttr = UA_VariableAttributes_default;
enAttr.dataType = UA_TYPES[UA_TYPES_EUINFORMATION].typeId;
enAttr.userWriteMask = UA_ACCESSLEVELMASK_READ;
UA_Variant_setScalarCopy(&enAttr.value, eu, &UA_TYPES[UA_TYPES_EUINFORMATION]);

UA_Server_addVariableNode(server, UA_NODEID_NULL, new_node_id, UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
								new_qualified_name, UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), enAttr, NULL,
								&temp_units);
UA_Server_addReference(server, temp_units, UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
							 UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), true);

if (retval != UA_STATUSCODE_GOOD) {
	UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%s: %s", new_qualified_name.name.data,
					UA_StatusCode_name(retval));
}

return new_node_id;
}
```

GoetzGoerisch, 2022-06-29:
@ccvca you have done this on our Sample Server, right?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier