open62541/open62541 #4601: how to read OPC UA events and alarms using  QT OPC UA  client?

Saleem344, 2021-08-17:
Hi,
I want read events from QT OPC UA client but but I couldn't find any example code how to read it from QT 
can anyone help me ??

Thanks your help.

jpfr, 2021-08-18:
https://github.com/open62541/open62541/blob/master/examples/tutorial_client_events.c

Does that help?

Saleem344, 2021-08-18:
> https://github.com/open62541/open62541/blob/master/examples/tutorial_client_events.c
> 
> Does that help?

Hi @jpfr ,
what you have shared is a direct example of C programming but I'm looking for similar example with QT integration.

QT OPC UA is open62541 API as back-end 

I want to read OPC UA Events from QT OPC UA client, I have found (https://doc.qt.io/QtOPCUA/qopcuamonitoringparameters-eventfilter.html) in the QT OPC UA documention but there is example how to subscribe to this events and where do I get the result of events ??

Can you help me with this??

jpfr, 2021-08-18:
Th qt wrapper is not maintained here.
You may ask the friendly folks at basyskom.
@basyskom-meerkoetter

closed 2021-08-22 by jpfr