open62541/open62541 #4619: Browse server for node with certain browsename

Psoke, 2021-08-30:
## Description
Hello there,

I have a question regarding the browse process. Somewhere in the "nodetree" of my server is a node with the browsename "FINDME". I have the browsename and I need the nodeID. How can I do that?
So far I managed to find the node and get its nodeID IF I have the nodeID of its parent (here 2, 5003), but I would like to keep the function as dynamic as possible. I used the code below from one of the open62541 examples.

`int findNodeId(UA_Client *client, char *browsename, int *nodeidnumeric)
{
	UA_BrowseRequest BrowseReq;
	UA_BrowseRequest_init(&BrowseReq);
	BrowseReq.requestedMaxReferencesPerNode = 0;
	BrowseReq.nodesToBrowse = UA_BrowseDescription_new();
	BrowseReq.nodesToBrowseSize = 1;
	BrowseReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(2, 5003);
	BrowseReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL;
	UA_BrowseResponse BrowseResp = UA_Client_Service_browse(client, BrowseReq);

	for (size_t i = 0; i < BrowseResp.resultsSize; ++i)
	{
		for (size_t j = 0; j < BrowseResp.results[i].referencesSize; ++j)
		{
			UA_ReferenceDescription *ref = &(BrowseResp.results[i].references[j]);
			if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC)
			{
				if (strcmp(browsename, ref->browseName.name.data) == 0)
				{
					*nodeidnumeric = ref->nodeId.nodeId.identifier.numeric;
				}
			}
		}
	}
	UA_BrowseRequest_clear(&BrowseReq);
	UA_BrowseResponse_clear(&BrowseResp);
	return EXIT_SUCCESS;
}`

Do I need to list all nodes on every layer, access them one by one recursively till I find the one containing the node "FINDME"?

thx for your time
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-08-30:
Do you know which “folder” contains findme or its type definition? Then you only need to call browse once...

Psoke, 2021-08-30:
We are setting up certain "Skills" that are offered network-wide and machine-independently, but the exact position in each servers graph might vary.

Lets say on every server there is a folder "Skills" that holds the nodes that I am searching for by name. The nodes like "FINDME" would be direct children of the folder "Skills". Would that help me?

If I would like to browse the server node by node, is there a termination condition I could use when I explored a "branch" to its fullest? The only thing I noticed is that the size of the returned nodes is exactly 1 when I can not get further "down".

jpfr, 2021-08-30:
Both options would help.
You can use the service Browse or TranslateBrowsePathToNodeId to then resolve the actual NodeId.

Psoke, 2021-08-30:
But I don’t know where exactly in the server node graph they will be placed, and they certainly will be placed differently on different servers.

I don’t really see how the afore mentioned helps me with navigating the graph as I still don’t know on which condition to break the browse loop?

I see what you mean but for Browse or TranslateBrowsePathToNodeId to work I would still need the exact placement of the node I am looking for itself or it’s parent? Or am I mistaken?

jpfr, 2021-09-10:
If you use specific object- or variabletype, you can follow the references from there.
Otherwise you might have to do recursive searching.
Why can’t you put nodes into a similar place?

closed 2021-09-16 by jpfr