open62541/open62541 #4622: Question about namespace indexes used in the tutorials

JensMertelmeyer, 2021-09-01:
## Question
I have a question which is probably terribly stupid.

In "[Working with objects and types](https://open62541.org/doc/current/tutorial_server_object.html)", and "[Working with variable types](https://open62541.org/doc/current/tutorial_server_variabletype.html)" it is shown how to add variable nodes and objects as children to, for example, the "objects" folder (`UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER)`). The supplied UA_NodeId is `UA_NODEID_NULL`, so open62541 will pick a numeric node id in the same namespace (zero).

This is what I understand, and looking at the returned node ids in the debugger confirms that.

However, in the tutorial "[Adding variables to a server](https://open62541.org/doc/current/tutorial_server_variable.html)", the method `UA_Server_addVariableNode` is called with an explicit requested node id for the new node: `UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");`

I am confused by two things:
1. Why is this, suddenly, namespace **1** instead of **0**?
2. Why is it called "integerNodeId"? The name suggests a numeric node id whereas it is, in fact, a string.

## Bonus Chatter
OPC clients from Softing, Prosys or UaExpert have no problems if I define the new node IDs in either namespace 0 or 1.
However, I have a client with some "Siemens Simatic S-7" whose software doesn't even display the node IDs for the variables (namespace = 1), but does so for the folders (namespace = 0).
[image]
I believe I should put the variables in namespace 0 instead?

I am no expert on OPC UA, to me, it sounded like I should always use namespace 1:

> Namespace for Nodes defined in the local Server. This may include types and instances used in a Device represented by the Server. This namespace shall have namespace index 1.

Source: https://reference.opcfoundation.org/v104/DI/v102/docs/11.2/

 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Other OPC UA SDKs used (client or server): Siemens Simatic S-7
 - [x] Operating system: Windows x64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Unfortunately, no
 - [ ] Wireshark network dump attached: Unfortunately, no
 - [x] Self-contained code example attached
 - [ ] Critical issue: Not yet 😎

ccvca, 2021-09-02:
Dear @JensMertelmeyer 
as you already read in the specification, Namespace 0 is not the right namespace  for custom objects in the address space. I think this has been choosen, as `UA_NODEID_NULL` was the only way in the past to get any unused nodeid. In the recent versions you can get an unused Nodeid of any Namespace (`nsIndex`) by using `UA_NODEID_NUMERIC(nsIndex, 0)`.

If you are working on a production-ready OPC UA Server, I would recomment to use a custom dedicated Namespace for all of your nodes.
```c
UA_UInt16 nsIdx = UA_Server_addNamespace(server, "http://example.com/UA/MachineAlpha");
// nsIndex contains the index of your namespace. Use this variable in all places, where you need the namespace index.
```

JensMertelmeyer, 2021-09-02:
Thank you very much, that was most helpful.

I will try adding everything to a new namespace. So node id **as well as browse name** should go into a new namespace index (most likely 2)?

Not sure if this is out of scope, but can we also make pull requests for the documentation, especially the tutorial? Or make formal suggestions? I only found raw html files, and I'm not sure pull requests on those are going to helpful.

ccvca, 2021-09-02:
Aside from the nodes (which should all be in your additional namespace), the browsenames may have the index of your namespace only if the browse name is specified by you. For BrowseNames of members, the index is the same as in the type definition. (See: https://reference.opcfoundation.org/Core/docs/Part3/6.4.1/ )

JensMertelmeyer, 2021-09-03:
I moved all the node ids for type definitions, objects and variables to a completely new namespace (index = 2) - Including the qualified names.

I have confirmation from my client that he is now able to read everything with this Siemens S7-thing. 😊

Thank you very much, @ccvca

closed 2021-09-03 by JensMertelmeyer