open62541/open62541 #4623: writing string array from open62541 client to open62541 server

embbo, 2021-09-02:
## Description

   UA_Variant input = {NULL};
    const size_t array_size = 2;
    UA_String* values = (UA_String*)UA_Array_new(array_size, &UA_TYPES[UA_TYPES_STRING]);
   
    values[0] = UA_String_fromChars("1");
    values[1] = UA_String_fromChars("2");
    values[2] = UA_String_fromChars("3");

    UA_Variant_setArray(&input, values, array_size, &UA_TYPES[UA_TYPES_STRING]);
    size_t outputSize;
    UA_Variant* output;
    retval = UA_Client_call(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
        UA_NODEID_NUMERIC(1, 7777), 1, &input, &outputSize, &output);
    if (retval == UA_STATUSCODE_GOOD) {
        printf("Method call was successful, and %lu returned values available.\n",
            (unsigned long)outputSize);
        UA_Array_delete(output, outputSize, &UA_TYPES[UA_TYPES_VARIANT]);
    }
    else {
        printf("Method call was unsuccessful, and %x returned values available.\n", retval);
    }
    UA_Client_delete(client); /* Disconnects the client internally */
    return EXIT_SUCCESS;

but the other side I get garbage values like this 

Server:

1y840ts  
2us- 
3pen6254 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

JensMertelmeyer, 2021-09-03:
Most probably not the main cause of your problem, but shouldn't `array_size` have a value of `3`, not `2`?

embbo, 2021-09-15:
@JensMertelmeyer No its no an issue  just trying to do different things--

o6-nbeier, 2026-10-20:
Thank you for the report.

A `UA_String` is length-delimited rather than NUL-terminated, so pass one Variant holding an array of 3 Strings (arraySize 3 in a single input argument) and read it length-aware in the method callback.

Closing for now.

closed 2026-10-20 by o6-nbeier