open62541/open62541 #4635: /plugins/mqtt/ua_mqtt_adapter.c : connectMqtt() : trying to understand the logic

jackybek, 2021-09-11:
## Description

## Background Information / Reproduction Steps
In the first half of this function block, it make use of mqttCaFilePath/ mqttCaPath, mqttClientCertPath & mqttClientKeyPath to create a '**SSL context**' object. At the end of this segment, i did not get any error with this function call:

     long verifyResult = SSL_get_verify_result(channelData->ssl);

Further down the function block, it created a client object <**client**> of type struct mqtt_client and update mqttClientId, mqttClientUsername & mqttClientPassword fields.  Then it tries to connect. using 

      mqttErr = mqtt_connect(client, clientId, NULL, NULL, 0, username, password, 0, 400);

I get this message on the screen:

       [2021-09-11 10:31:51.921 (UTC+0800)] info/server        PubSub MQTT Connection established.

However, when the program runs to the point where it calls yieldMqtt(), it fails with the following error message:

       open62541.c : yieldMqtt(): error: MQTT_ERROR_SOCKET_ERROR

After much thoughts about it, i suspect it is due to the following:

1. in the command prompt, to publish a message to the broker i tested this successfully : 

       mosquitto_pub -u < uname > -P <12345> -h 192.168.1.11 -p 8883 -t AirgardTopic -m "My Message" --capath /etc/ssl/certs --cafile /etc/ssl/certs/mosq-ca.crt --cert /etc/ssl/certs/mosq-client-115.crt --key /etc/ssl/certs/mosq-client-115.key --tls-version tlsv1.2 -d -i OPCServer-115

2.  However, in the client object <**client**>, only clientid (OPCServer-115), username( uname ) and password(12345) is set.  There is no way to set the following required parameters:
     mqttCaFilePath
     mqttCaPath, 
     mqttClientCertPath
     mqttClientKeyPath

How does the created '**SSL context**' object interact with the client object to pass in the 4 required parameters?

3. I suspect that because these 4 parameters are missing, when trying to connect to the mqtt broker, it failed.

Please advise if my understanding is correct.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): v1.2.2-434-gf52a7938-dirty
 - [ X] Other OPC UA SDKs used (client or server): openssl v1.1.1L
 - [ X] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-jvoe, 2021-09-24:
All TLS related settings are not of any interest to the MQTT client.
The TLS connection is opened in ua_mqtt_adapter.c and passed to the client here: https://github.com/open62541/open62541/blob/master/plugins/mqtt/ua_mqtt_adapter.c#L308

I don't see where you are setting the mqttUseTLS option. I'd guess that all the errors you have described in your bug reports are caused by accidentally opening a non-TLS connection on a port where TLS is required.

jackybek, 2021-09-25:
I configured port 8884 on the mqtt server, and open port 8884 as the client.

jackybek, 2021-09-26:
For better clarity, let me share my code and mqtt config:

# 1. mosquitto.conf
#Global settings
per_listener_settings true
allow_zero_length_clientid true
max_connections 1024
message_size_limit 1024000
#max_keepalive 65535 : no such parameter

#Default listener - plain MQTT protocol
#MQTT, unencrypted, unauthenticated
port 1883
require_certificate false
protocol mqtt
allow_anonymous false
password_file /etc/mosquitto/conf.d/userlist.txt

#MQTT, unencrypted, authenticated
listener 1884
require_certificate false
allow_anonymous false
protocol mqtt
password_file /etc/mosquitto/conf.d/userlist.txt

#MQTT, encrypted (over TLS/SSL), unauthenticated
#listener 8883
#require_certificate false
#allow_anonymous true
#protocol mqtt
#capath /etc/ssl/certs/
#cafile /etc/ssl/certs/mosq-ca.crt
#password_file /etc/mosquitto/conf.d/userlist.txt
#tls_version tlsv1.1

#MQTT, encrypted (over TLS/SSL), client certificate required
listener 8884
require_certificate true
password_file /etc/mosquitto/conf.d/userlist.txt
protocol mqtt
capath /etc/ssl/certs/
cafile /etc/ssl/certs/mosq-ca.crt
#tls_version tlsv1.1

# 2. C source code

        #define CA_FILE_PATH_MQTT             "/etc/ssl/certs/mosq-ca.crt"           
        #define CA_PATH_MQTT                           "/etc/ssl/certs"
        #define CLIENT_CERT_PATH_MQTT                  "/etc/ssl/certs/mosq-client-115.crt"   
        #define CLIENT_KEY_PATH_MQTT                   "/etc/ssl/certs/mosq-client-115.key"   

        //--connectionOptionsIndex start at 0
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, CONNECTIONOPTION_NAME_MQTT);   // mqttClientId
        UA_String mqttClientId = UA_STRING(CLIENT_ID_MQTT);
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &mqttClientId, &UA_TYPES[UA_TYPES_STRING]);

        // preallocate sendBufferSize
        connectionOptionsIndex++;       // 1
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "sendBufferSize");
        UA_UInt32 sendBufferSize = 32767;       // must be UA_UInt32 to match open62541.c codes
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &sendBufferSize, &UA_TYPES[UA_TYPES_UINT32]);

        // preallocate recvBufferSize
        connectionOptionsIndex++;       // 2
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "recvBufferSize");
        UA_Int32 recvBufferSize = 32767;        // must be UA_UInt32 to match open62541.c codes
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &recvBufferSize, &UA_TYPES[UA_TYPES_UINT32]);

        connectionOptionsIndex++;       // 3
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, USERNAME_OPTION_NAME_MQTT);    // mqttUsername
[... 47 lines cut ...]
      Client Mqtt-11 sending SUBSCRIBE (Mid: 1, Topic: AirgardTopicEdgeToCloud, QoS: 0)
      Client Mqtt-11 received SUBACK
      Subscribed (mid: 1): 0

        2. mosquitto_pub -u myuid -P mypwd -h 192.168.1.11 -p 8884 -t AirgardTopicEdgeToCloud -m "My Message" --capath /etc/ssl/certs --cafile /etc/ssl/certs/mosq-ca.crt --cert /home/pi/Downloads/openest.io/mosq-serv-11.crt --key /home/pi/Downloads/openest.io/mosq-serv-11.key -d -i Mqtt-11 --tls-version tlsv1.2

I get this response:
Client Mqtt-11 sending CONNECT
Client Mqtt-11 received CONNACK (0)
Client Mqtt-11 sending PUBLISH (d0, q0, r0, m1, 'AirgardTopicEdgeToCloud', ... (10 bytes))
Client Mqtt-11 sending DISCONNECT

Whenever i run mosquitto_pub, i will get this on (1):
Client Mqtt-11 sending CONNECT
Client Mqtt-11 received CONNACK (0)
Client Mqtt-11 sending SUBSCRIBE (Mid: **2**, Topic: AirgardTopicEdgeToCloud, QoS: 0)
Client Mqtt-11 received SUBACK
Subscribed (mid: **2**): 0

where the number 2 will increment by 1 every time i run mosquitto_pub

Since it is proven that certificates for MQTT server and client is valid and the parameters are in order, i now matched it to my source code.  Unfortunately, i always get the following error :
open62541.c : yieldMqtt() 214841 (/plugins/ua_network_pubsub_mqtt.c) : error: MQTT_ERROR_SOCKET_ERROR

Is there a working example that I can verify my code against so as to resolve the above errors?

jackybek, 2021-09-26:
I simulated a test and this is the outcome:

1. run the mqtt broker : 

         mosquitto_sub -u myuid-P mypwd -h 192.168.1.11 -p 8884 -t AirgardTopicEdgeToCloud --capath /etc/ssl/certs --cafile /etc/ssl/certs/mosq-ca.crt --cert /home/pi/Downloads/openest.io/mosq-serv-11.crt --key /home/pi/Downloads/openest.io/mosq-serv-11.key -d -i Mqtt-11 --tls-version tlsv1.1

2. run OPCServer with mosquitto client port set as 8884 (T+0)

          212430 msg->state : 1 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE

3.  after running for 15 sec, a message arrive on the broker at T+15.  This is what is see in OPCUA Server:

          212430 msg->state : 1 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
          212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE

4.  after running for another 12 sec, another message arrive on the broker at T+27.

         212430 msg->state : 1 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE

5.  In the next cycle around T+40, OPCUA server sends another message to the broker.  This time round, I cannot see the message arriving at the broker, but there is no error on the OPCUA server.

         212430 msg->state : 1 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE

6. In the next cycle around T+ 53, OPCUA server sends another message and now it fail with this message

         212430 msg->state : 1 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 2 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         212430 msg->state : 0 (0 = MQTT_QUEUED_UNSENT, 1 = MQTT_QUEUED_AWAITING_ACK, 2 = MQTT_QUEUED_COMPLETE
         214838 YieldMqtt() : mqtt_sync return value : 0x80000011
         [2021-09-26 12:02:14.160 (UTC+0800)] error/server       open62541.c : yieldMqtt() 214841 (/plugins/ua_network_pubsub_mqtt.c) : error: MQTT_ERROR_SOCKET_ERROR

So, from a code perspective, it is quite obvious that the handshake between OPCUA server and MQTT broker is successful initially.
However, why did the publishing routine fail after 2 rounds of successful attempts baffles me.

Please advise.

jackybek, 2021-09-26:
I dig deeper and found the trigger.  It is in mqtt_pal_sendall().
For the first 2 successful cycle, the return value of SSL_write is > 0.
For the next cycle, the return value of SSL_write is <=0, so it went into the case section.   The error return from SSL_get_error(ssl, rv) = 5.  I then look up /usr/include/openssl/ssl.h and 5 means SSL_ERROR_SYSCALL.

Because the switch segment does not handle SSL_ERROR_SYSCALL, it return the default value which is set as MQTT_ERROR_SOCKET_ERROR;

From openssl:
SSL_ERROR_SYSCALL
Some I/O error occurred. The OpenSSL error queue may contain more information on the error. If the error queue is empty (i.e. ERR_get_error() returns 0), ret can be used to find out more about the error: If ret == 0, an EOF was observed that violates the protocol. If ret == -1, the underlying BIO reported an I/O error (for socket I/O on Unix systems, consult errno for details).

I then inspect ERR_get_error() when i get SSL_ERROR_SYSCALL.  It return 0, pointing to this explanation: **an EOF was observed that violates the protocol.**

                switch (error) {
                case SSL_ERROR_WANT_READ:               // 2
                case SSL_ERROR_WANT_WRITE:              // 3
                case SSL_ERROR_WANT_ACCEPT:             // 8
                case SSL_ERROR_WANT_CONNECT:            // 7
                case SSL_ERROR_WANT_X509_LOOKUP:        // 4
                case SSL_ERROR_ZERO_RETURN:             // 6
                    return written;
                case SSL_ERROR_SYSCALL:         // 5
                        printf("ERR_get_error()for case = SSL_ERROR_SYSCALL is %d \n", ERR_get_error());

basyskom-jvoe, 2021-09-27:
That's interesting, I never saw that error during development.

During development of the TLS support, I extended https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_mqtt_publish.c and tested against the public Mosquitto broker mentioned in the comment header.

jackybek, 2021-09-28:
I am testing mosquitto mqtt using port 8884 with client side certificate and keys. 

Can I check which port were you testing with testmosqitto.org ?

jackybek, 2025-11-20:
closed due to revamped event loop.