open62541/open62541 #4641: Segmentation Fault in UA_Server_addPubSubConnection() function working with pubsub

alipdri, 2021-09-12:
## Description
I am writing a c++ wrapper to work with open62541 and I have created a ```PubSubConfig``` class that allocates and stores the pubsub config types e.g. ```UA_PubSubConnectionConfig, UA_PublishedDataSetConfig etc...``` as its member and returns references to them with getter functions. My application compiles whiteout errors and is simply creating a new connection with ```UA_ServerConfig_setMinimal``` and tries to add a pubsub connection to it in the ```UA_PubSubTransportLayerUDPMP()``` transport layer (following the sample codes provided in `examples/pubsub/tutorial_pubsub_subscribe.c`). 

The problem is that I am receiving these two errors when I run my application:

```
error/server	PubSub Connection creation problem.Preparing the socket multicast configuration failed
error/server	PubSub Connection creation failed. Transport layer creation problem.
```

and when i try to run with debugger i get a ```SegFault``` with the following stack:
```
1  msvcrt!memmove                                                 0x7ff94ac14452 
2  UA_Array_copy                  ua_types.c                 1362 0x405d34       
3  String_copy                    ua_types.c                 152  0x402d45       
4  copyStructure                  ua_types.c                 1100 0x40557b       
5  UA_copy                        ua_types.c                 1217 0x4058de       
6  UA_Array_copy                  ua_types.c                 1370 0x405d79       
7  Variant_copy                   ua_types.c                 608  0x403fc3       
8  UA_copy                        ua_types.c                 1217 0x4058de       
9  UA_Variant_copy                types_generated_handling.h 745  0x40f3fa       
10 UA_PubSubConnectionConfig_copy ua_pubsub_writer.c         43   0x40f7e5       
11 UA_Server_addPubSubConnection  ua_pubsub_manager.c        46   0x416dd0       
12 Server::addPubSubConnection    server.cpp                 65   0x401bf1       
13 Server::preparePubSub          server.cpp                 26   0x401942       
14 main
```

which as I examined it further is happening when function `UA_PubSubConnectionConfig_copy` tries to copy the pubsub connection configs to a `tmpConnectionConfig` variable and results in a violation in a `memcpy` that is copying two pointers.

## Background Information / Reproduction Steps

Here is the link to my project shared on gitlab and the steps to build and reproduce the error:

```
git clone https://gitlab.com/Paridari/opcua.git <dir>
cd <dir>
cmake .
make
./Gateway
```

Used CMake options:
```bash
cmake -DUA_ENABLE_PUBSUB:BOOL=ON
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2.2-422-gcde8c716
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Windows 10 / 64 bits
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

liing0228, 2022-03-19:
HI @alipdri :
Did you sove the problem?
i face the same problem now
[image]

Zeke0302, 2022-11-04:
Hi @liing0228 
I also face the same problem.
please change your publisher example UA_NetworkAddressUrlDataType structure value.

UA_NetworkAddressUrlDataType networkAddressUrlSub = {UA_STRING_NULL , UA_STRING("opc.udp://224.0.0.22:62541/")};