open62541/open62541 #4652: Make binary encoding part of public api

mlgiraud, 2021-09-17:
## Description
Currently the encoding and decoding functions are not part of the public api of the stack. I propose adding these to the public api. In some cases users might want to directly encode opcua datatypes into a bytestring and send them this way. An example would be the TrustListType. The read and write methods use a ByteString as return value and argument. This bytestring is an encoded TrustListDataType object. The functions to encode and decode this object already exist in the stack or are automatically generated, but simply not part of the public api.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): N/A
 - [ ] Operating system: N/A
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached N/A
 - [ ] Wireshark network dump attached N/A
 - [ ] Self-contained code example attached N/A
 - [ ] Critical issue

jpfr, 2021-09-19:
Yes. Important point.
This was done for binary and json encoding.

see #4597

ok to close?

mlgiraud, 2021-09-20:
Yes this works. I didn't realize this was already implemented last month. Sorry.
Although i have one thing to add: Wouldn't it be better to also expose the helper functions to decode or encode each type directly as well? I.e. expose all `UA_<TYPE>_<encode|decode>Binary` functions. This seems a bit more intuitive than going directly over the UA_encodeBinary function via the types array.

jpfr, 2021-09-20:
That is fairly easy to add to the python script that generates the existing handling functions for all types.

Given that the encoding is only used in very few places, lets not add the overhead at this time.
How many usages are we talking? More than a dozen?

Once this gets inconvenient, we can easily add it.

mlgiraud, 2021-09-21:
Sounds reasonable. I will open a PR if it reaches a point where it gets too inconvenient. Closing for now.

closed 2021-09-21 by mlgiraud