open62541/open62541 #4653: CTT Issue: Subscription Transfer Test 001.js fails

phuong-nm, 2021-09-17:
## Description

server_ctt without UA_ENABLE_ENCRYPTION fails CTT subscription transfer test 001.js. 

The server_ctt example program was built as-is using open62541 master branch @17.09.2021.

## Background Information / Reproduction Steps

"This TransferSubscription Service is used to transfer a Subscription and its MonitoredItems from one Session to another.
The Server shall validate that the Client of that Session is operating on behalf of the same user and that the potentially new Client supports the Profiles that are necessary for the Subscription." [OPCUA Part 4, 5.13.7 ](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.7/)

The CTT test verifies transferring a subscription from a closed session to a new one. Both sessions are created by the same user.
The first session is closed with DeleteSubscription set to FALSE, causing the subscription to be detached from the session e.g., relationship between subscription and related session is delete, see below.

```c
void
UA_Session_detachSubscription(UA_Server *server, UA_Session *session,
                              UA_Subscription *sub) {
    /* Detach from the session */
    sub->session = NULL;
    TAILQ_REMOVE(&session->subscriptions, sub, sessionListEntry);

    /* ... */
}
```

Afterwards the client, who created the subscription in the first session, reconnects and requests to transfer the subscription to new session. Since the subscription is already detached from session, the server lost information required to verify if transferring of the subscription is allowed as described in the specification "_The Server shall validate that the Client of that Session is operating on behalf of the same user_".

```c
static void
Operation_TransferSubscription(UA_Server *server, UA_Session *session,
                               const UA_Boolean *sendInitialValues,
                               const UA_UInt32 *subscriptionId,
                               UA_TransferResult *result) {
    /* Get the subscription. This requires a server-wide lookup instead of the
     * usual session-wide lookup. */
    UA_Subscription *sub = UA_Server_getSubscriptionById(server, *subscriptionId);
    if(!sub) {
        result->statusCode = UA_STATUSCODE_BADSUBSCRIPTIONIDINVALID;
        return;
    }

    /* Is this the same session? Return the sequence numbers and do nothing else. */
    UA_Session *oldSession = sub->session; // <----- HERE: oldSession = NULL
    if(oldSession == session) {
        result->statusCode = setTransferredSequenceNumbers(sub, result);
        return;
    }
 
    /* Check with AccessControl if the transfer is allowed */
    if(!server->config.accessControl.allowTransferSubscription ||
       !server->config.accessControl.
       allowTransferSubscription(server, &server->config.accessControl,
                                 oldSession ? &oldSession->sessionId : NULL,
                                 oldSession ? oldSession->sessionHandle : NULL,
                                 &session->sessionId, session->sessionHandle)) {
        result->statusCode = UA_STATUSCODE_BADUSERACCESSDENIED;
        return;
    }

    /* ... */
}
```

andreasebner, 2025-02-04:
The mentioned code has been revised and new ctt runs pass. If you still have problems, please create a new issue.

closed 2025-02-04 by andreasebner

mlang-de, 2025-02-04:
Imho at least for branch 1.3 this is still an open problem.