open62541/open62541 #4660: Question: How to correctly add a custom array variable to an object type definition?

JensMertelmeyer, 2021-09-21:
## Question
I would like to define an object type that has a mandatory variable of a float array.
What I do works, but I am not sure if I'm doing it correctly, it feels wrong to me:

1. I call `addDataTypeNode` for adding a new subtype of float, calling it `MyFloat`
2. I call `addVariableTypeNode` for adding a new subtype of the BaseDataVariableType, calling it `MyArray`. I set a default value of a float array (10 items). As a type definition, I explicitly pass the NodeId of `MyFloat`
3. I call `addObjectTypeNode` to add `MyObject` as a subtype of BaseObjectType
4. I try to use `addVariableNode` to add a variable of type `MyArray` to `MyObject` but that's where I stumble

Step 2 (adding the type `MyArray`) seems useless because I have no idea how I could use it in step 4.
Basically, I am just settings an array of float as a value, not an instance of `MyArray`.

## Background Information / Reproduction Steps

Full code below, here is the gist of it:
```cpp
int main(int argc, char **argv) {
	server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addFloatAlias();        // step 1
	addArrayType();         // step 2
	addObjectType();        // step 3
	addArrayToObjectType(); // step 4
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}
```
My step 4 looks like this:

```cpp
void addArrayToObjectType() {
	// Intention: Every instance of "MyObject" should have
	// have a mandatory variable "TheArray" of type "MyArray"

	UA_VariableAttributes va = UA_VariableAttributes_default;
	// I expected I would just have to do this:
	va.dataType = MyArray;
	
	// Instead, I have to do this:	
	const size_t COUNT = 5; 
	va.dataType = MyFloat;
	va.valueRank = UA_VALUERANK_ONE_DIMENSION;
	va.arrayDimensionsSize = 1;
	UA_UInt32 arrayDims[1] = {COUNT};
	va.arrayDimensions = arrayDims;
	va.arrayDimensionsSize = 1;
	UA_Float zero[COUNT] = {};
	UA_Variant_setArray(&va.value, zero, COUNT, &UA_TYPES[UA_TYPES_FLOAT]);
	// I thought the default value of "MyArray" was already clear
	// Instead, I'm basically setting the value to an array of "MyFloat"
	// instead of leveraging the type "MyArray"

	UA_Server_addVariableNode(
		server,
		UA_NODEID_NULL,
		MyObject,
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
		UA_QUALIFIEDNAME_ALLOC(0, "TheArray"),
		/*typeDefinition:*/ MyArray,
		va,
		nullptr,
		&MyObject_TheArray
	);
}
```

## Full code example
<details>
  <summary>Here is the complete code (113 lines)</summary>

```cpp
#include <open62541/server.h>
#include <open62541/server_config_default.h>

void addFloatAlias();
void addArrayType();
void addObjectType();
void addArrayToObjectType();

UA_Server *server;
UA_NodeId MyFloat, MyArray, MyObject, MyObject_TheArray;

int main(int argc, char **argv) {
	server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);
	
	addFloatAlias();        // step 1
	addArrayType();         // step 2
	addObjectType();        // step 3
	addArrayToObjectType(); // step 4
	
	bool *isRunning = new bool(true);
	UA_Server_run(server, isRunning);
}

void addFloatAlias() {
	UA_DataTypeAttributes dta = UA_DataTypeAttributes_default;
	UA_Server_addDataTypeNode(
		server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_FLOAT),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		UA_QUALIFIEDNAME_ALLOC(0, "MyFloat"),
		dta,
		nullptr,
		&MyFloat
	);
}

void addArrayType()  {
	const size_t COUNT = 10;
	UA_VariableTypeAttributes vtAttr = UA_VariableTypeAttributes_default;
	vtAttr.dataType = MyFloat;
	vtAttr.valueRank = UA_VALUERANK_ONE_DIMENSION;
	UA_UInt32 arrayDims[1] = {COUNT};
	vtAttr.arrayDimensions = arrayDims;
	vtAttr.arrayDimensionsSize = 1;

	/* a matching default value is required: Let's make it ten zeroes */
	UA_Float zero[COUNT] = {};
	UA_Variant_setArray(&vtAttr.value, zero, COUNT, &UA_TYPES[UA_TYPES_FLOAT]);

	UA_Server_addVariableTypeNode(
		server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		UA_QUALIFIEDNAME_ALLOC(0, "MyArray"), 
		MyFloat, // no idea what this actually means
		vtAttr,
		nullptr,
		&MyArray
	);
}

void addObjectType() {
	UA_ObjectTypeAttributes ota = UA_ObjectTypeAttributes_default;
	UA_Server_addObjectTypeNode(
		server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		UA_QUALIFIEDNAME_ALLOC(0, "MyObject"),
		ota,
		nullptr,
		&MyObject
	);
}

void addArrayToObjectType() {
	// Intention: Every instance of "MyObject" should have
	// have a mandatory variable "TheArray" of type "MyArray"

	UA_VariableAttributes va = UA_VariableAttributes_default;
	// I expected I would just have to do this:
	va.dataType = MyArray;
	
	// Instead, I have to do this:	
	const size_t COUNT = 5; 
	va.dataType = MyFloat;
	va.valueRank = UA_VALUERANK_ONE_DIMENSION;
	va.arrayDimensionsSize = 1;
	UA_UInt32 arrayDims[1] = {COUNT};
	va.arrayDimensions = arrayDims;
	va.arrayDimensionsSize = 1;
	UA_Float zero[COUNT] = {};
	UA_Variant_setArray(&va.value, zero, COUNT, &UA_TYPES[UA_TYPES_FLOAT]);
	// I thought the default value of "MyArray" was already clear
	// Instead, I'm basically setting the value to an array of "MyFloat"
	// instead of leveraging the type "MyArray"

	UA_Server_addVariableNode(
		server,
		UA_NODEID_NULL,
		MyObject,
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
		UA_QUALIFIEDNAME_ALLOC(0, "TheArray"),
		/*typeDefinition:*/ MyArray,
		va,
		nullptr,
		&MyObject_TheArray
	);
}
```
</details>

## Bonus Chatter
I found the example "custom data types" which consists of the files 
1. [custom_datatype.h](https://github.com/open62541/open62541/blob/13212bdde3514af0fb8732afa0a8329bb9cdbfb5/examples/custom_datatype/custom_datatype.h)
2. [server_types_custom.c](https://github.com/open62541/open62541/blob/13212bdde3514af0fb8732afa0a8329bb9cdbfb5/examples/custom_datatype/server_types_custom.c)

This is much more complex and requires setting up binary encoding of a struct. It looks even "cooler", and I am not sure if this is the way it should usually be done. My doubt is that this is too complex for my clients to read, they usually use rather simple PLCs like a Siemens S-7.

## Checklist

Please provide the following information:

 - [x] open62541 Version **1.2**
 - [x] Other OPC UA SDKs used (client or server): **[UnifiedAutomation UaExpert](https://www.unified-automation.com/products/development-tools/uaexpert.html)**
 - [x] Operating system: **Windows x64**
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: **None needed**
 - [x] Wireshark network dump attached: **Not needed**
 - [x] Self-contained code example attached: **Yes**
 - [x] Critical issue: **No**

jpfr, 2021-09-30:
This is already nearly correct.

you now only have to make the variable member mandatory. Via a special reference. See the examples.

also, if you define the array dimensions in the variable type, they are a hard requirement for instances.
And created on the fly if not present for a new variable.

o6-nbeier, 2026-10-20:
Closing for now.

closed 2026-10-20 by o6-nbeier