open62541/open62541 #4741: Log message "Server url is invalid: %.*s" causes segmentation fault when connecting to empty URL

pettineh, 2021-11-02:
## Description
Calling _UA_Client_connect_ function with completely empty address seems to cause a segmentation fault in the logging implementation. When the logger gets a message with content: "Server url is invalid: %.*s", the segmentation fault occurs when the va_list variables are being accessed. My guess is that the char pointer pointing to the message url is invalid when it is passed to the logger. 

## Background Information / Reproduction Steps
Now all this happens when the address given to _UA_Client_connect_ function is completely empty null-terminated char array. Adding even a single whitespace character to the array fixes the issue.

To reproduce the issue, these are the lines of code I execute inside my OPC UA client implementing class:
        UA_Client* client = UA_Client_new();
        UA_ClientConfig* config = UA_Client_getConfig(client);
        UA_ClientConfig_setDefault(config);
        config->logger.log = &myLogHandler;
        config->logger.context = this;
        char url[] = "";
        UA_Client_connect(client, &url); // Causes error log message: "Server url is invalid: %.*s"

Inside myLogHandler callback I call the _strlen_ for the first variable returned by the va_list object holding the string arguments. This causes segfault when the log message is the one mentioned above.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Other OPC UA SDKs used (client or server): client
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

kimim, 2021-11-04:
compilation warning on my machine, maybe you need to pass `url` other than `&url` here.

```
tutorial_client_firststeps.c: In function 'main':
tutorial_client_firststeps.c:19:54: warning: passing argument 2 of 'UA_Client_connect' from incompatible pointer type [-Wincompatible-pointer-types]
   19 |     UA_StatusCode retval = UA_Client_connect(client, &url);
      |                                                      ^~~~
      |                                                      |
      |                                                      char (*)[1]
In file included from tutorial_client_firststeps.c:11:
open62541.h:28385:50: note: expected 'const char *' but argument is of type 'char (*)[1]'
28385 | UA_Client_connect(UA_Client *client, const char *endpointUrl);
      |                                      ~~~~~~~~~~~~^~~~~~~~~~~

```

pettineh, 2021-11-04:
> compilation warning on my machine, maybe you need to pass `url` other than `&url` here.
>
> ```
> tutorial_client_firststeps.c: In function 'main':
> tutorial_client_firststeps.c:19:54: warning: passing [...]

My bad. For the code example I simplified the code a bit. I originally passed the const char * to the function. I will edit the code example in the original post.

kimim, 2021-11-05:
If everything is fine, could you close this issue?

pettineh, 2021-11-09:
@kimim Now you misunderstood, the issue still exists. The code example I gave was just bad.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier