open62541/open62541 #4743: Adding variable to server failed (tutorial_server_variable.c).

kwitekrac, 2021-11-02:
## Description
The issue is connected with adding variables to the server. I have tried to learn how to do it by using:
- tutorial_server_variable.c
- the example from the main open62541 github page

in both cases, I got an error when the variable is added.

## Background Information / Reproduction Steps
I'm using open62541/1.3 and have tried two setups:
### Windows 8.1, build with visual studio 2019

Used CMake commandline options:
-DUA_ENABLE_AMALGAMATION:BOOL="1" -DBUILD_SHARED_LIBS:BOOL="1" -DCMAKE_BUILD_TYPE:STRING="Release" 

I copied the dll and header file to folder with tutorial_server_firststeps.c (#include open62541.h used) and compile it with: 

`gcc -std=c99 myServer.c -l"ws2_32" -L. -lopen62541 -o myServer`

This example works. Then I tried to do the same with tutorial_server_variable.c, the server log is: 
>[2021-11-02 13:46:15.328 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-11-02 13:46:15.329 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2021-11-02 13:46:15.329 (UTC+0100)] info/server        AccessControl: x509 certificate user authentication is enabled
[2021-11-02 13:46:15.329 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-11-02 13:46:15.329 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-11-02 13:46:15.330 (UTC+0100)] info/session       SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNode: Node could not create a node
with error code BadNodeAttributesInvalid
[2021-11-02 13:46:15.420 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://lukasz_laptop:4840/

### Ubuntu 20.04.3
In this case, to build a library I follow this tutorial:
https://opcua.rocks/step-1-setup-open62541-on-linux-build-first-server/
the only difference is I used amalgamation and build_shared_libs.

The tutorial_server_variable.c not worked also this time, log:
>lukasz@lukasz-VirtualBox:~/serverVariable$ ./myServer
[2021-11-02 11:48:56.614 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-11-02 11:48:56.614 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2021-11-02 11:48:56.614 (UTC+0100)] info/server	AccessControl: x509 certificate user authentication is enabled
[2021-11-02 11:48:56.615 (UTC+0100)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-11-02 11:48:56.615 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-11-02 11:48:56.615 (UTC+0100)] info/session	SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadTypeMismatch
Writing a string returned statuscode BadTypeMismatch
[2021-11-02 11:48:56.615 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://lukasz-VirtualBox:4840/
^C[2021-11-02 12:51:36.249 (UTC+0100)] info/server	received ctrl-c
[2021-11-02 12:51:36.249 (UTC+0100)] info/network	Shutting down the TCP network layer

kimim, 2021-11-04:
Hello,

I think this is expected result, that the tutorial code write a string to an integer variable, so the log says `BadTypeMismatch`

see the code:

```C
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myString, &UA_TYPES[UA_TYPES_STRING]);
    UA_StatusCode retval = UA_Server_writeValue(server, myIntegerNodeId, myVar);
    printf("Writing a string returned statuscode %s\n", UA_StatusCode_name(retval));
```

kimim, 2021-11-04:
by the way, you do not need to specify `-DBUILD_SHARED_LIBS:BOOL="1"` for the tutorial.
see the article I wrote: http://kimi.im/2021-11-04-open62541-tutorials-01

kwitekrac, 2021-11-04:
Thank you for your help. The ubuntu error code could be expected however, the windows log is different - so the project build may be different than that one on Linux. 
Shared lib is what I want to use and just need to check if it is working. 

The only difference during the build on windows is that I didn't provide tls.mbed (just build on Windows fails), but I didn't enable UA_ENABLE_ENCRYPTION. I just wonder to know if that could be connected with adding node problem in example with varaible.

closed 2024-09-23 by kwitekrac