open62541/open62541 #4745: How to compile the new open65421 on qnx6.5

wangyuan02605, 2021-11-03:
I cannot cross compile open65421 in the current version or previous versions

I execute $ cmake -DCMAKE_TOOLCHAIN_FILE= ../tools/cmake/Toolchain-QNX-6.5.0.cmake .. -G "Unix Makefiles"

$mak
[image]
e

wangyuan02605, 2021-11-03:
add execuate in 1.2.2
[image]

kimim, 2021-11-04:
seems you are still using win32 arch, maybe you need to specify `-DUA_ARCHITECTURE=posix` for `cmake`

jpfr, 2021-11-04:
Try setting UA_ARCHITECTURE to `posix`.
You get errors for including win32 system APIs. That won't work on QNX.

closed 2021-11-27 by wangyuan02605