open62541/open62541 #4754: Function to check Channel State

AlejandroAntonio1998, 2021-11-05:
Hi everyone!

I want to implement a function to **check the status** of my **communication channel**. The code is as follows:

`void checkComunicacion (UA_Client* &Cliente_OPC_UA) {

	UA_SecureChannelState* channelState;
	UA_Client_getState(Cliente_OPC_UA, channelState, NULL, NULL);

	if (*channelState== UA_SECURECHANNELSTATE_OPEN) {
		cout << "OPEN" << endl;
	}
}`

If I write directly the code after starting the connection it works fine, but if I call the code through the function it doesn´t work.

Does anyone have an idea about what can be the problem?

Thanks!

kimim, 2021-11-05:
I define this:
```C
void check_connection(UA_Client* &ua_client) {
    UA_SecureChannelState* channelState;
    UA_Client_getState(ua_client, channelState, NULL, NULL);

    if (*channelState == UA_SECURECHANNELSTATE_OPEN) {
      printf("OPEN\n");
    }
}
```
and call it:

```C
    check_connection(client);
```

Works fine.

AlejandroAntonio1998, 2021-11-08:
Hi Kimim!
I´ve tried your code and works fine if I generate the connection 'directly', but if I use the function below and later I call check_connection it cracks.

Function to start communication (there are some global variables used):
```
bool Set_Communication (UA_Client* &Client_OPC_UA) {

	UA_Client* client = UA_Client_new();
	UA_ClientConfig* cc = UA_Client_getConfig(client);

	UA_ByteString* cert = loadFile(client_cert);
	UA_ByteString* key = loadFile(client_key);

	size_t trustListSize = 1;
	UA_STACKARRAY(UA_ByteString*, trustList, trustListSize+1);
	trustList[0] = loadFile(server_cert);

	UA_ClientConfig_setDefaultEncryption(cc, *cert, *key, *trustList, trustListSize, NULL, 0);

	cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	cc->clientDescription.applicationUri = UA_STRING_ALLOC(app_URI);
	cc->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

	cc->timeout = 5 * 1000; 
	cc->secureChannelLifeTime = 15 * 60 * 1000;
	cc->requestedSessionTimeout  = 30 * 60 * 1000 ; 
	cc->connectivityCheckInterval = 0;

	UA_StatusCode status;
	status = UA_Client_connectUsername(client, server_address, username, password);
	if(status != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return false;
	}

        Client_OPC_UA = client;

	return true;
}
```
I´ve used it to read and write variables from a server. The problem seems to be that channelState is not written when I call check_connection. Do you have an idea about what could be the problem?

Thanks!!!

kimim, 2021-11-08:
Sorry, in my previous reply the function definition is not correct, how about change it from:
```C
bool Set_Communication (UA_Client* &Client_OPC_UA) {
```

```C
bool Set_Communication (UA_Client* Client_OPC_UA) {
```

AlejandroAntonio1998, 2021-11-09:
It doesn´t work :(

Thanks anyway !!!

kimim, 2021-11-11:
sorry there is to limited information.
Could you add more description, how is your code structure, what is defined and what is called later?

jpfr, 2021-11-20:
Try this adaptation of the code that was posted:

```c
void checkComunicacion (UA_Client* &Cliente_OPC_UA) {

UA_SecureChannelState channelState;
UA_Client_getState(Cliente_OPC_UA, &channelState, NULL, NULL);

if (*channelState== UA_SECURECHANNELSTATE_OPEN) {
	cout << "OPEN" << endl;
}
}
```

Note that I call with the address to an existing variable.
Not an undefined pointer going nowhere.

AlejandroAntonio1998, 2021-11-29:
Hi, I´ve changed the code as you told me and it works fine.
Thank you so much!!!!