open62541/open62541 #4761: UA_String_fromChars() not null terminating the string

vsaljooghi, 2021-11-10:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

In file: src/ua_types.c
Function: UA_String UA_String_fromChars(const char *src)

When an input string is copied, it will be copied up to strlen() of the string and unfortunately it will not be null terminated. If the person is lucky, the memory has already zero values otherwise it will not work.

andreasebner, 2021-11-11:
Strings are not null terminated in OPC UA (in compliance to the standard). The String and Bytestring DataTypes are structures with a poiunter to your chars and the length information. The function reads the input char array until there is a null termination and copys the string into an UA_Byte array and set the length.

jpfr, 2021-11-19:
As documented, the input string has to be null-terminated.
The output string is a "fat pointer" that has an explicit length and is not null-terminated.

closed 2021-11-19 by jpfr