open62541/open62541 #4762: FreeLibrary does not unload dll

skubidu1324, 2021-11-10:
## Description
As part of a larger project we have one open62541 server thread (c++). We load the dll with `LoadLibrary` and unload it with `FreeLibrary` but can't unload the dll after calling `UA_ServerConfig_setMinimal`.

## Background Information / Reproduction Steps
The thread works with `UA_Server_run_startup(server)` and  `UA_Server_run_iterate(server, true)` in the worker process. At the end `UA_Server_run_shutdown(server)` and  `UA_Server_delete(server)`. The log said 'Shutting down the TCP network layer', everthing seems to be good, but `FreeLibrary` does not unload the dll because some other handle has loaded it, but we don't know who it is. 

When I  only do this
```
UA_Server* server = UA_Server_new();
UA_ServerConfig* config = UA_Server_getConfig(server);
UA_ServerConfig_setMinimal(config, 4840, NULL);
UA_Server_delete(server);
```

I get the same behavior.  `UA_ServerConfig_setMinimal` does something, so I can't unload my dll. If I commented out the `UA_ServerConfig_setMinimal` line  it works fine and the dll can be unloaded. Does someone have a clue what to do to truely free my dll? 

## Checklist

Please provide the following information:

 - [ x ] open62541 Version (release number or git tag): 1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x ] Operating system: Win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-11-17:
Your minimal example doesn't do much.
It won't even open a socket.

We do however call WSAStartup() to enable networking.
That internally loads further dlls.

Try to manually call WSACleanup. That will free networking resources and also unload dlls.

Sadly, there is no good place to have WSACleanup in our code.
As there could be several server instances in the same process...

jpfr, 2021-11-18:
Scratch that. We do call WSACleanup internally.

skubidu1324, 2021-11-18:
Yes, stopping the `UA_ServerNetworkLayer` is calling  `WSACleanup()`. Do you have another idea?

skubidu1324, 2021-11-18:
I found it! In `UA_OpenSSL_LoadLocalCertificate` you have to call `OPENSSL_free(derData)` in case of else. Now I can free my dll.

closed 2021-11-19 by skubidu1324