open62541/open62541 #4763: Installation issue: Make do not end after 12h stuck on "Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o"

rfri4s, 2021-11-10:
## Description

I executed the make with open62541 the version 1.2.2 on a Raspberry PI 2 B 32-bits running "Raspbian GNU/Linux 10 (buster)" but after 12 hours the make stills in step: "Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o" 

## Background Information / Reproduction Steps

I run this commands
cd /tmp/
git clone https://github.com/open62541/open62541.git 
cd open62541/
git submodule update --init --recursive 
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
make

## Logs

pi@raspberrypi:/tmp/open62541/build $ cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.16")
-- Found Git: /usr/bin/git (found version "2.20.1")
-- open62541 Version: 1.0.6
-- The selected architecture is: posix
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Success
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2
-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2 - Success
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE)
-- Could NOT find LATEX (missing: LATEX_COMPILER)
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/open62541/build
pi@raspberrypi:/tmp/open62541/build $ make
Scanning dependencies of target open62541-generator-types
[  1%] Generating src_generated/open62541/types_generated.c, src_generated/open62541/types_generated.h, src_generated/open62541/types_generated_handling.h, src_generated/open62541/types_generated_encoding_binary.h
[  1%] Built target open62541-generator-types
Scanning dependencies of target open62541-generator-namespace
[  3%] Generating src_generated/open62541/namespace0_generated.c, src_generated/open62541/namespace0_generated.h
INFO:__main__:Preprocessing /tmp/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
[  3%] Built target open62541-generator-namespace
Scanning dependencies of target open62541-generator-statuscode
[  5%] Generating src_generated/open62541/nodeids.h
[  5%] Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c
[  5%] Built target open62541-generator-statuscode
Scanning dependencies of target open62541-generator-transport
[  7%] Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h, src_generated/open62541/transport_generated_encoding_binary.h
[  7%] Built target open62541-generator-transport
Scanning dependencies of target open62541-object
[  9%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[  9%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 11%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/types_generated.c.o
[ 13%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/transport_generated.c.o
[ 13%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/statuscodes.c.o
[ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_util.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_workqueue.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_timer.c.o
[ 19%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 21%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 23%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session.c.o
[ 23%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o
[ 25%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_ns0.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_config.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o
[ 30%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_utils.c.o
[ 30%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_discovery.c.o
[ 32%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_securechannel_manager.c.o
[ 34%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session_manager.c.o
[ 34%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_networkmessage.c.o
[ 36%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_manager.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_ns0.c.o
[ 40%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o
[ 42%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_method.c.o
[ 42%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o
[ 44%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o
[ 46%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o
[ 48%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o
[ 48%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_monitoreditem.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o
[ 51%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o
[ 51%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery_multicast.c.o
[ 53%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client.c.o
[ 55%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_connect.c.o
[ 55%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_connect_async.c.o
[ 57%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_discovery.c.o
[ 59%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o
[ 59%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_subscriptions.c.o
[ 61%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_worker.c.o
[ 63%] Building C object CMakeFiles/open62541-object.dir/deps/libc_time.c.o
[ 63%] Building C object CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o
[ 65%] Building C object CMakeFiles/open62541-object.dir/deps/base64.c.o
[ 67%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o

12hours later:

^C^Cmake[2]: *** Deleting file 'CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o'
make[2]: *** [CMakeFiles/open62541-object.dir/build.make:694: CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:358: CMakeFiles/open62541-object.dir/all] Interrupt
make: *** [Makefile:152: all] Interrupt
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2.2
 - [ ] Other OPC UA SDKs used (client or server): NA
 - [ ] Operating system: Raspbian GNU/Linux 10 (buster)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

rfri4s, 2021-11-10:
I did the same test with the 
Raspberry Pi 3 Model B Rev 1.2
Raspbian GNU/Linux 10 (buster)

Same problem

rfri4s, 2021-11-10:
+ Download links on https://open62541.org/ do not work

mlgiraud, 2021-12-15:
The Problem is most likely that the namespace0 file is very large and requires considerable processing power to compile. I would suggest you cross-compile the library on a more powerful device than the raspberry pi.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier