open62541/open62541 #4764: msvc2008 headers compatibility

toninlg, 2021-11-10:
## Description

I'm stuck with msvc2008. I managed to build open62541 with it by forcing C++ compilation but in order to use it I cannot force C++ compilation and some headers are not compatible:
- open62541\plugin\log.h
- open62541\util.h
- open62541\client.h
- open62541\plugin\nodestore.h

Is it possible to keep headers compliant with C standard (declaration of local variables at top of the functions)?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -G "Visual Studio 9 2008" -DUA_ENABLE_ENCRYPTION:BOOL="0" -DUA_FORCE_CPP:BOOL="1" -DBUILD_SHARED_LIBS:BOOL="1" -DOPEN62541_VERSION:STRING="v1.2.2" -S . -B ./build
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-11-13:
> Is it possible to keep headers compliant with C standard (declaration of local variables at top of the functions)?

No. That's too much effort at this point.
Newer versions of MSVC can build just fine.

And for MSVC 2008 the C++ mode works.
See the MSVC 2008 build on the appveyor infrastructure.

closed 2021-11-13 by jpfr

toninlg, 2021-11-17:
I'm surprised it's too much work. I'm only speaking about the headers who are not compatible, not the code: with the cmake options used it's only four headers.