open62541/open62541 #4765: OpenSSL 3.0 compilation errors

toninlg, 2021-11-10:
## Description

I have compilation errors with OpenSSL 3.0 because some functions are deprecated and a pointer has been changed to const one. 

```
9>..\plugins\crypto\openssl\securitypolicy_openssl_common.c(211) : warning C4996: 'RSA_size': Since OpenSSL 3.0
9>        C:\Program Files (x86)\OpenSSL-Win32\include\openssl/rsa.h(204) : see declaration of 'RSA_size'
9>..\plugins\crypto\openssl\securitypolicy_openssl_common.c(211) : warning C4996: 'EVP_PKEY_get0_RSA': Since OpenSSL 3.0
9>        C:\Program Files (x86)\OpenSSL-Win32\include\openssl/evp.h(1346) : see declaration of 'EVP_PKEY_get0_RSA'
9>..\plugins\crypto\openssl\securitypolicy_openssl_common.c(223) : error C2664: 'RSA_private_decrypt' : cannot convert parameter 4 from 'const rsa_st *' to 'RSA *'
9>        Conversion loses qualifiers
9>..\plugins\crypto\openssl\securitypolicy_openssl_common.c(282) : error C2440: '=' : cannot convert from 'const rsa_st *' to 'RSA *'
9>        Conversion loses qualifiers
9>..\plugins\crypto\openssl\securitypolicy_openssl_common.c(443) : error C2440: 'initializing' : cannot convert from 'const rsa_st *' to 'RSA *'
```
https://www.openssl.org/docs/man3.0/man7/migration_guide.html

> ### Functions that return an internal key should be treated as read only
> 
> Functions such as EVP_PKEY_get0_RSA(3) behave slightly differently in OpenSSL 3.0. Previously they returned a pointer to the low-level key used internally by libcrypto. From OpenSSL 3.0 this key may now be held in a provider. Calling these functions will only return a handle on the internal key where the EVP_PKEY was constructed using this key in the first place, for example using a function or macro such as EVP_PKEY_assign_RSA(3), EVP_PKEY_set1_RSA(3), etc. Where the EVP_PKEY holds a provider managed key, then these functions now return a cached copy of the key. Changes to the internal provider key that take place after the first time the cached key is accessed will not be reflected back in the cached copy. Similarly any changes made to the cached copy by application code will not be reflected back in the internal provider key.
> 
> For the above reasons the keys returned from these functions should typically be treated as read-only. To emphasise this the value returned from EVP_PKEY_get0_RSA(3), EVP_PKEY_get0_DSA(3), EVP_PKEY_get0_EC_KEY(3) and EVP_PKEY_get0_DH(3) have been made const. This may break some existing code. Applications broken by this change should be modified. The preferred solution is to refactor the code to avoid the use of these deprecated functions. Failing this the code should be modified to use a const pointer instead. The EVP_PKEY_get1_RSA(3), EVP_PKEY_get1_DSA(3), EVP_PKEY_get1_EC_KEY(3) and EVP_PKEY_get1_DH(3) functions continue to return a non-const pointer to enable them to be "freed". However they should also be treated as read-only.
> 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -G "Visual Studio 9 2008" -DUA_ENABLE_ENCRYPTION:BOOL="1" -DUA_ENABLE_ENCRYPTION_OPENSSL:BOOL="1" -DUA_FORCE_CPP:BOOL="1" -DOPEN62541_VERSION:STRING="v1.2.2" -DOPENSSL_ROOT_DIR="C:/Program Files (x86)/OpenSSL-Win32/" -S . -B ./build
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

JohannesKauffmann, 2023-05-25:
Since [v1.3.4](https://github.com/open62541/open62541/releases/tag/v1.3.4), OpenSSL 3.0 support has been added (#5349). Otherwise, disable warnings as errors.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier