open62541/open62541 #4766: Unable to access server with User an Password (UA_Client_connectUsername)

POWER-KI, 2021-11-11:
# Description
I can't access the server with UA_Client_connectUsername

# Background Information / Reproduction Steps

Used CMake options:

Commandline options:
-DUA_FILE_NS0:BOOL="0" -DUA_FORCE_32BIT:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="REDUCED" -DUA_ENABLE_ENCRYPTION:BOOL="1" -DUA_ENABLE_ENCRYPTION_MBEDTLS:BOOL="1" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DUA_ENABLE_DISCOVERY:BOOL="1" -DMBEDCRYPTO_LIBRARY:FILEPATH="P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib" -DMAKEINDEX_COMPILER:FILEPATH="MAKEINDEX_COMPILER-NOTFOUND" -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL="0" -DMBEDTLS_INCLUDE_DIRS:PATH="P:/XU/PWK/WIN/MBEDTLS/include" -DMBEDTLS_LIBRARY:FILEPATH="P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib" -DUA_MSVC_FORCE_STATIC_CRT:BOOL="0" -DMBEDX509_LIBRARY:FILEPATH="P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib" -DUA_AMALGAMATION_ARCHITECTURES:STRING="" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="0" 

UA_FILE_NS0:BOOL=0
UA_FORCE_32BIT:BOOL=1
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_ENABLE_ENCRYPTION:BOOL=1
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=1
UA_ENABLE_AMALGAMATION:BOOL=1
UA_ENABLE_DISCOVERY:BOOL=1
MBEDCRYPTO_LIBRARY:FILEPATH=P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=0
MBEDTLS_INCLUDE_DIRS:PATH=P:/XU/PWK/WIN/MBEDTLS/include
MBEDTLS_LIBRARY:FILEPATH=P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib
UA_MSVC_FORCE_STATIC_CRT:BOOL=0
MBEDX509_LIBRARY:FILEPATH=P:/XU/PWK/WIN/MBEDTLS/visualc/VS2010/Release/mbedTLS.lib
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=0

# CODE to test
```
int TEST_UA()
{
	// Load certificate generated by UA-Expert
	UA_ByteString* certificate = loadFileUA("uaexpert.der");
	UA_ByteString* privateKey = loadFileUA("uaexpert_key.der");

	// Trust list;
	size_t trustListSize = 0;

	UA_STACKARRAY(UA_ByteString, trustList, trustListSize);

	// Revocation list
	UA_ByteString* revocationList = NULL;
	size_t revocationListSize = 0;

	// Config
	UA_Client* client = UA_Client_new();
	UA_ClientConfig* config = UA_Client_getConfig(client);
	config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	UA_ClientConfig_setDefaultEncryption(config, *certificate, *privateKey,
		trustList, trustListSize,
		revocationList, revocationListSize);

	// Set description as specified in certificate
	config->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", "UaExpert");
	config->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;
	config->clientDescription.applicationUri = UA_STRING_ALLOC("urn:WIN10-01:UnifiedAutomation:UaExpert");

	// Connect
	
	UA_StatusCode retval = UA_Client_connectUsername(client, "opc.tcp://192.168.2.4:48010", "joe", "god");	
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return 0;		
	}
return 1;
}
```
Return code Error 0x80020000

# Checklist

 - [ ] open62541 Version (1.2.2):
 - [ ] Other OPC UA SDKs used (client ): 
 - [ ] Win10:

#Thank You

kimim, 2021-11-11:
the error code is `UA_STATUSCODE_BADINTERNALERROR`
maybe there is some network connection errors.

POWER-KI, 2021-11-11:
If I use  UA_Client_connect(client, "opc.tcp://192.168.2.4:48010"); 
it works

POWER-KI, 2021-11-11:
From Log:
```
AcceptAll Certificate Verification. Any remote certificate will be accepted.
The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
Connection 2772 | SecureChannel 256546819 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
Rejecting endpoint 0: security mode doesn't match
Rejecting endpoint 1: security mode doesn't match
Rejecting endpoint 2: security mode doesn't match
Rejecting endpoint 3: security mode doesn't match
No suitable endpoint found
Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError

```

POWER-KI, 2021-11-13:
This is my solution to the age-old problem of connecting to UA-EXPERT Server with User and Password.
1) you should have the CERTIFICATE end its KEY to do so , if you have chosen to use MBED TSL you can follow the instructions in this [guide ](https://tls.mbed.org/kb/how-to/generate-a-self-signed-certificate) 
(Note. the content of the certificate in this case does not matter);
2) Do not define any config->securityMode.

My working code:
```
UA_ByteString* loadFileUA(const char* path)
{
	FILE* f = fopen(path, "rb");
	if (f == NULL)
		return NULL;

	fseek(f, 0, SEEK_END);
	long fsize = ftell(f);
	fseek(f, 0, SEEK_SET);  /* same as rewind(f); */

	UA_ByteString* result = UA_ByteString_new();
	UA_ByteString_allocBuffer(result, (size_t)fsize + 1);
	memset(result->data, 0, result->length);
	fread(result->data, result->length, 1, f);
	fclose(f);

	return result;
}

int TEST_UA()
{
	// Load your certificate
	UA_ByteString* certificate = loadFileUA("XPLAB_CERT.der");
	UA_ByteString* privateKey = loadFileUA("XPLAB_KEY.der");

	// Trust list;
	size_t trustListSize = 0;

	UA_STACKARRAY(UA_ByteString, trustList, trustListSize);

	// Revocation list
	UA_ByteString* revocationList = NULL;
	size_t revocationListSize = 0;

	// Config
	UA_Client* client = UA_Client_new();
	UA_ClientConfig* config = UA_Client_getConfig(client);
	//config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;  !! NOT DEFINE this	
	UA_ClientConfig_setDefaultEncryption(config, *certificate, *privateKey,
		trustList, trustListSize,
		revocationList, revocationListSize);

	// Set description as specified in certificate !!NOT NECESSARY
	//config->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", "UaExpert");
	//config->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;
	//config->clientDescription.applicationUri = UA_STRING_ALLOC("urn:WIN10-01:UnifiedAutomation:UaExpert");

	// Connect
	
	UA_StatusCode retval = UA_Client_connectUsername(client, "opc.tcp://192.168.X.X:48010", "joe", "god");	
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return 0;		
	}
return 1;
}

```
NOTE 1:
if you apply the standard config  UA_ClientConfig_setDefault and after UA_ClientConfig_setDefaultEncryption
for some reasons it does not works.

NOTE 2:
we are updating POWER-KI the code adding the certificate, key pair to the OPC_NEW function:

```
opc=OPC_NEW(£CLI,NULL,"XPLAB_CERT.der", "XPLAB_KEY.der");
°r=OPC_CLI(opc,£OPEN,"opc.tcp://192.168.X.X:48010","joe","god");
```