open62541/open62541 #4770: open62541 1.2.2 based server cannot be connected to on kernel 3.17

tboeckel, 2021-11-12:
## Description
Since we upgraded to open62541 1.2.2 from 1.0.1 servers running on an old 3.17 Linux kernel refuse any incoming connections due to an invalid type identifier 12208.

UaExpert just logs these two error messages:
Discovery FindServersOnNetwork on opc.tcp://192.168.20.175:4840 failed (BadServiceUnsupported), falling back to FindServers
Discarding Server iNOEX ATUM OPC-UA server with URL

The open62541 based server on the other side gives more details about the failed connection:
[2021-11-12 19:31:06.753 (UTC+0000)] info/network	Connection 26 | New connection over TCP from 10.10.11.2
[2021-11-12 19:31:06.760 (UTC+0000)] info/channel	Creating a new SecureChannel
[2021-11-12 19:31:06.820 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | Opened SecureChannel
[2021-11-12 19:31:06.870 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | Unknown request with type identifier 12208
[2021-11-12 19:31:06.921 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | CloseSecureChannel
[2021-11-12 19:31:06.921 (UTC+0000)] info/network	Connection 26 | Processing the message failed with error BadConnectionClosed
[2021-11-12 19:31:06.921 (UTC+0000)] info/network	Connection 26 | Closed
[2021-11-12 19:31:06.963 (UTC+0000)] info/network	Connection 26 | New connection over TCP from 10.10.11.2
[2021-11-12 19:31:06.970 (UTC+0000)] info/channel	Creating a new SecureChannel
[2021-11-12 19:31:07.015 (UTC+0000)] info/channel	Connection 26 | SecureChannel 2 | Opened SecureChannel
[2021-11-12 19:31:07.124 (UTC+0000)] info/channel	Connection 26 | SecureChannel 2 | CloseSecureChannel
[2021-11-12 19:31:07.125 (UTC+0000)] info/network	Connection 26 | Processing the message failed with error BadConnectionClosed
[2021-11-12 19:31:07.125 (UTC+0000)] info/network	Connection 26 | Closed

However, the same binary running on a system with a slightly more recent kernel 4.14 can be successfully connected to. The only difference of that Linux system is the kernel. All binaries of the operation system are identical between both systems. Only the kernels differ.

Back with open62541 1.0.x the connection succeeded even with kernel 3.17. Something between open62541 1.0.x and open62541 1.2.x must have been changed in respect to incoming server connections which makes a substantial difference between different Linux kernel versions.

Unfortunately we cannot update the systems running the 3.17 kernel to a more recent kernel version. Hence we need a fix in open62541, if possible. open62541 1.2.2 offers many features, like asynchronous method handling, that our software relies on by now.

The quite puzzling detail is that UaExpert is able to obtain the server's description string, because the error message of UaExpert contains the description text "iNOEX ATUM OPC-UA server" which definitely must have been sent out from our server.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.0.1 and 1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux with kernel 3.17 and kernel 4.14
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2021-11-12:
This is a slightly more verbose log created with UA_LOGLEVEL=0

[2021-11-12 20:27:34.354 (UTC+0000)] trace/network	Connection 26 | New TCP connection on server socket 25
[2021-11-12 20:27:34.355 (UTC+0000)] info/network	Connection 26 | New connection over TCP from 10.10.11.2
[2021-11-12 20:27:34.361 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.361 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.362 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Process a HEL message
[2021-11-12 20:27:34.405 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.405 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.406 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Verifying chunk signature
[2021-11-12 20:27:34.406 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Process an OPN message
[2021-11-12 20:27:34.406 (UTC+0000)] debug/channel	Connection 26 | SecureChannel 1 | Generating new local nonce
[2021-11-12 20:27:34.406 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2021-11-12 20:27:34.457 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.458 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.458 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 1 | Generating new local keys
[2021-11-12 20:27:34.458 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 1 | Generating new remote keys
[2021-11-12 20:27:34.458 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 1 | Process a MSG
[2021-11-12 20:27:34.458 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | Unknown request with type identifier 12208
[2021-11-12 20:27:34.458 (UTC+0000)] debug/server	Sending response for RequestId 2 with ServiceResult BadServiceUnsupported
[2021-11-12 20:27:34.510 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.510 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.510 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 1 | Process a CLO
[2021-11-12 20:27:34.510 (UTC+0000)] info/channel	Connection 26 | SecureChannel 1 | CloseSecureChannel
[2021-11-12 20:27:34.510 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.510 (UTC+0000)] info/network	Connection 26 | Closed
[2021-11-12 20:27:34.559 (UTC+0000)] trace/network	Connection 26 | New TCP connection on server socket 25
[2021-11-12 20:27:34.560 (UTC+0000)] info/network	Connection 26 | New connection over TCP from 10.10.11.2
[2021-11-12 20:27:34.560 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.561 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.561 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Process a HEL message
[2021-11-12 20:27:34.611 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.612 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.612 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Verifying chunk signature
[2021-11-12 20:27:34.612 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 0 | Process an OPN message
[2021-11-12 20:27:34.612 (UTC+0000)] debug/channel	Connection 26 | SecureChannel 2 | Generating new local nonce
[2021-11-12 20:27:34.612 (UTC+0000)] info/channel	Connection 26 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2021-11-12 20:27:34.663 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.663 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.664 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 2 | Generating new local keys
[2021-11-12 20:27:34.664 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 2 | Generating new remote keys
[2021-11-12 20:27:34.664 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 2 | Process a MSG
[2021-11-12 20:27:34.664 (UTC+0000)] debug/session	SecureChannel 2 | Session g=00000000-0000-0000-0000-000000000000 | Processing FindServersRequest
[2021-11-12 20:27:34.664 (UTC+0000)] debug/session	SecureChannel 2 | Session g=00000000-0000-0000-0000-000000000000 | Sending response for RequestId 2 of type FindServersResponse
[2021-11-12 20:27:34.723 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.723 (UTC+0000)] trace/network	Connection 26 | Received a packet.
[2021-11-12 20:27:34.723 (UTC+0000)] trace/channel	Connection 26 | SecureChannel 2 | Process a CLO
[2021-11-12 20:27:34.724 (UTC+0000)] info/channel	Connection 26 | SecureChannel 2 | CloseSecureChannel
[2021-11-12 20:27:34.726 (UTC+0000)] trace/network	Connection 26 | Activity on the socket
[2021-11-12 20:27:34.726 (UTC+0000)] info/network	Connection 26 | Closed

jpfr, 2021-11-17:
The trace looks quite normal.
The client calls FindServersRequest, then correctly winds down the connection.

Can you also extract logs from UAExpert?
There are config settings to write a log to a file.

It's possible that UAExpert is unhappy for some reason...

tboeckel, 2021-11-17:
UaExpert logs nothing else than the two lines mentioned in the original ticket.

However, in the meantime I found out what really makes the difference between the two systems, except the different kernel versions. Somehow the 3.17 kernel does not fully support IPv6. As soon as I restrict open62541 to incoming IPv4 connections only everything is working again for kernel 3.17 connections. As mentioned above already the more recent kernel 4.14 was never affected.

In essence all I changed in the ServerNetworkLayerTCP_start() function in arch/network_tcp.c is this:

#if UA_IPV6
     // allow IPv4 and IPv6
     hints.ai_family = AF_UNSPEC;
#else
     // enforce IPv4 only
     hints.ai_family = AF_INET;
#endif

If IPv6 is enabled, then IPv4 and IPv6 is allowed for a socket, otherwise only IPv4 is allowed. And the definition of UA_IPV6 depends on a private setting for certain systems, instead of enabling it unconditionally for POSIX systems.

I don't know whether it is reasonable to include this patch in open62541. Why allow non-IPv4 sockets, when IPv6 is not enabled (for open6541)?

So from my point of view the issue is fixed for me. The original issue is not the fault of open62541, but due to some weird setup for the 3.17 kernel, which was done by some other person than me. All I can say is that things changed between open62541 1.0.x and 1.1.x regarding IPv4 and IPv6 handling, because release 1.0.x worked on that strange 3.17 kernel.

To make it short: this issue can be closed. If you like, include my patch in future releases of open62541.

jpfr, 2021-11-21:
Thanks for looking into that.
The fix is in #4788. It goes into 1.2 first and trickles down to 1.3 and master.
Closing this.

closed 2021-11-21 by jpfr