open62541/open62541 #4774: Can't set source timestamp on a tag from an Object

jvoosten, 2021-11-13:
I have found a discrepancy between nodes that are created directly or being instantiated from an Object description: in the latter I cannot set the source timestamp: it is always "now". Either I am doing something wrong, don't understand OPC objects properly or there's a bug in the code, not sure...

I first tested the "server_settimestamp.c" example to see if the 'fake' source timestamp is set and read out correctly, and it is with UAExpert. I then extended the example with an OPC Object (following the "Pump" example from the Tutorial). Then create an instance of such an object and set its Status node. Unfortunately, although the code to set the variable is the same as for the fake timestamp, the status timestamp is always set to the "now". I can set the value and the OPC-Status, just not the timestamp(s).

Open62541 version: 1.2.1 git tag
Build options: UA_MULTITHREADING 100
OS: Windows 10
Compiler: Visual Studio CE 2019
Other OPC clients: UAExpert 1.4.4

Output log does not show much:

~~~
[2021-11-13 20:07:36.967 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-11-13 20:07:36.968 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2021-11-13 20:07:36.968 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-11-13 20:07:36.969 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-11-13 20:07:37.049 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://Zabimaru:4840/
[2021-11-13 20:07:41.391 (UTC+0100)] info/network       Connection 428 | New connection over TCP from 127.0.0.1
[2021-11-13 20:07:41.393 (UTC+0100)] info/channel       Connection 428 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-11-13 20:07:41.394 (UTC+0100)] warn/channel       Connection 428 | SecureChannel 1 | ActivateSession: Session not found
[2021-11-13 20:07:41.693 (UTC+0100)] info/channel       Connection 428 | SecureChannel 1 | Session 54ba8749-feb3-fbc3-3083-f84f4c4434e3 created
[2021-11-13 20:07:41.694 (UTC+0100)] info/session       SecureChannel 1 | Session ns=1;g=54ba8749-feb3-fbc3-3083-f84f4c4434e3 | ActivateSession: Session activated
[2021-11-13 20:07:41.705 (UTC+0100)] info/session       SecureChannel 1 | Session ns=1;g=54ba8749-feb3-fbc3-3083-f84f4c4434e3 | Subscription 1 | Created the Subscription with a publishing interval of 500.00 ms
[2021-11-13 20:07:41.706 (UTC+0100)] info/session       SecureChannel 1 | Session ns=1;g=54ba8749-feb3-fbc3-3083-f84f4c4434e3 | Subscription 1 | MonitoredItem 1 | Created the MonitoredItem (Sampling Interval: 250.000000ms, Queue Size: 0)
[2021-11-13 20:07:41.706 (UTC+0100)] info/session       SecureChannel 1 | Session ns=1;g=54ba8749-feb3-fbc3-3083-f84f4c4434e3 | Subscription 1 | MonitoredItem 2 | Created the MonitoredItem (Sampling Interval: 250.000000ms, Queue Size: 0)
~~~

Attached are a Wireshark capture and the source code I used to test. 

[opc-time-capture.zip](https://github.com/open62541/open62541/files/7532590/opc-time-capture.zip)
[examples_server_settimestamp.zip](https://github.com/open62541/open62541/files/7532591/examples_server_settimestamp.zip)

jpfr, 2021-11-15:
Good find. Thanks for the detailed and reproducible write-up.
The fix is in #4775.

jvoosten, 2021-11-16:
I needed to backport the fix to 1.2.2 since 1.3 crashes here (more about that in another ticket), but it solves the problem. Thanks for the quick fix!

closed 2021-11-16 by jvoosten