open62541/open62541 #4790: getUserAccessLevel documentation

kjeldflarup, 2021-11-22:
I can make my own getUserAccessLevel
   open62541_server_config->accessControl.getUserAccessLevel = my_getUserAccessLevel;

```
static UA_Byte 
my_getUserAccessLevel(UA_Server *server, UA_AccessControl *ac,
                                    const UA_NodeId *sessionId, void *sessionContext,
                                    const UA_NodeId *nodeId, void *nodeContext) {
    return 0xFF;
}
```

Now my question:

How do I use and control `sessionContext` and `nodeContext`?

I can see that the values are not set to NULL
```
thread 3 "dummycodesysser" hit Breakpoint 1, my_getUserAccessLevel (server=0x7ffff7f66edc <Array_decodeBinary+300>, ac=0x6, 
    sessionId=0x55555556c7b0 <UA_TYPES+144>, sessionContext=0x7ffff0001c05, nodeId=0x6, nodeContext=0x55555556c720 <UA_TYPES>) at ../opcuausers.c:176

```

I would expect that `nodeContext` is set when adding the node to the server, and likewise `sessionContext` when the user logs in.

jpfr, 2021-11-22:
Yes. You can set the session context in the same plugin.
In the activatesession callback.

kjeldflarup, 2021-11-23:
I can se that activateSession_default sets the session context to the username. I assume that this is not used elsewhere in open62541 it self.

kjeldflarup, 2021-11-23:
Writing my own  activatesession callback proves a bit cumbersome.

The structure `AccessControlContext` is declared in `plugins/ua_accesscontrol_default.c`  
This indicates, that `AccessControlContext` should not be used outside the open62541 core, as it may be subject to change without notice. 

To make this the correct way, the whole  `plugins/ua_accesscontrol_default.c`  should thus be cloned, which seems like a lot of extra work. And still vulnerable to changes in the core. 

Would it be possible to move  `AccessControlContext` to `plugin/accesscontrol_default.h`

kjeldflarup, 2021-11-23:
Never mind, the solution is to clone `plugins/ua_accesscontrol_default.c`

closed 2021-11-23 by kjeldflarup