open62541/open62541 #4792: Reconnection failure

toninlg, 2021-11-22:
## Description
On the same machine I connect open62541 as client to a server and add subscription with monitored items. I close the server then restart it.
The client tries to reconnect but fails with UA_STATUSCODE_BADSERVERHALTED 0x800E0000, then it says it's already connected while the return status of UA_Client_connect is still UA_STATUSCODE_BADSERVERHALTED.

```
while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, serverurl);
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                         "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

        UA_Client_run_iterate(client, 1000);
    };

```

```
[2021-11-22 15:47:53.350 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:54.353 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:55.356 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:56.359 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:57.362 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:57.377 (UTC+0100)] warn/client        Client already connected
[2021-11-22 15:47:57.779 (UTC+0100)] warn/channel       Connection 516 | SecureChannel 3 | Receiving the response failed with StatusCode BadConnectionClosed
[2021-11-22 15:47:57.784 (UTC+0100)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2021-11-22 15:47:57.784 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
Channel: Closed, Session: Created
[2021-11-22 15:47:57.784 (UTC+0100)] info/userland      The client is disconnected
[2021-11-22 15:48:00.011 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:00.011 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Created, ConnectStatus: BadDisconnect
Channel: Fresh, Session: Created
[2021-11-22 15:48:00.011 (UTC+0100)] info/userland      The client is disconnected
[2021-11-22 15:48:00.011 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:03.049 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:03.049 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:06.071 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:06.071 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:09.078 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:09.078 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:12.132 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:12.132 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:15.177 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:15.177 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:18.230 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:18.230 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:21.274 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:21.274 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:24.311 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:24.311 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:27.333 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.

[2021-11-22 15:48:27.333 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 15:48:30.355 (UTC+0100)] warn/network       Connection to opc.tcp://127.0.0.1:62640/IntegrationObjects/ServerSimulator failed with error: No connection could be made because the target machine actively refused it.
[... 93 lines cut ...]
[2021-11-22 16:05:02.509 (UTC+0100)] warn/client        Client already connected
[2021-11-22 16:05:02.509 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second
[2021-11-22 16:05:03.526 (UTC+0100)] warn/client        Client already connected
[2021-11-22 16:05:03.526 (UTC+0100)] error/userland     Not connected. Retrying to connect in 1 second

```
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -G "Visual Studio 9 2008" -DUA_ENABLE_ENCRYPTION:BOOL="1" -DUA_ENABLE_ENCRYPTION_OPENSSL:BOOL="1" -DUA_FORCE_CPP:BOOL="1" -DUA_FORCE_32BIT:BOOL="1" -DBUILD_SHARED_LIBS:BOOL="1" -DOPEN62541_VERSION:STRING="v1.2.2" -DOPENSSL_ROOT_DIR="C:/Program Files (x86)/OpenSSL-Win32-1.1.1/" -S . -B ./build
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

toninlg, 2021-12-05:
Nobody encounters same problem?

robert-johnston, 2022-08-10:
I have the same problem with essentially the same code example.  Open62541 v1.2.2-366-g4d4e206e.

I connect to a OPC UA server.  
Then shutdown the server.
Then restart the server.

The warn/client message from UA_Client_connect() call  repeats as "Client already connected"
The retval from the UA_Client_connect() call repeats as "BadServerHalted" 0x800E0000
And the loop continues without ever connecting, retval never comes back as Good.

Calling UA_Client_connect() does not reconnect even though the server is running.

lukasberbuer, 2023-05-12:
This is still an issue in v1.3.5: The example `client_subscription_loop.c` does not work as expected (as described above).
Are there any new best practises to handle disconnects with clients?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier