open62541/open62541 #4793: Error during compilation: variable "may be used uninitialized"

Mebus, 2021-11-22:
Hallo,

I am trying to compile the code on the tag v1.2.2 with the following cmake configuration:

```
 cmake -G "MinGW Makefiles" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DCMAKE_INSTALL_PREFIX="D:\Project\opcuaclientexperiment\libopen62541" -D UA_ENABLE_AMALGAMATION=TRUE -DUA_ENABLE_ENCRYPTION=TRUE -DUA_ENABLE_ENCRYPTION_OPENSSL=TRUE 
``` 

I am using Msys2 on Windows 10. Unfortunately I geht this error:

```
Integrating file 'D:/Project/open62541/plugins/crypto/ua_securitypolicy_none.c'...done.
Integrating file 'D:/Project/open62541/arch/win32/ua_clock.c'...done.
Integrating file 'D:/Project/open62541/arch/win32/ua_architecture_functions.c'...done.
Integrating file 'D:/Project/open62541/arch/network_tcp.c'...done.
The size of D:/Project/open62541/open62541.c is 10613856 Bytes.
[ 60%] Built target open62541-amalgamation-source
[ 60%] Built target open62541-code-generation
[ 65%] Building C object CMakeFiles/open62541-object.dir/open62541.c.obj
D:\Project\open62541\open62541.c: In function 'namespace0_generated':
D:\Project\open62541\open62541.c:177180:15: note: variable tracking size limit exceeded with '-fvar-tracking-assignments', retrying without
177180 | UA_StatusCode namespace0_generated(UA_Server *server) {
       |               ^~~~~~~~~~~~~~~~~~~~
D:\Project\open62541\open62541.c: In function 'UA_CertificateVerification_VerifyApplicationURI':
D:\Project\open62541\open62541.c:192801:9: error: 'subjectURI.length' may be used uninitialized [-Werror=maybe-uninitialized]
192801 |     if (UA_Bstrstr (subjectURI.data, subjectURI.length,
       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
192802 |                     applicationURI->data, applicationURI->length) == NULL) {
       |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Project\open62541\open62541.c:192763:27: note: 'subjectURI' declared here
192763 |     UA_String             subjectURI;
       |                           ^~~~~~~~~~
D:\Project\open62541\open62541.c:192801:31: error: 'subjectURI.data' may be used uninitialized [-Werror=maybe-uninitialized]
192801 |     if (UA_Bstrstr (subjectURI.data, subjectURI.length,
       |                     ~~~~~~~~~~^~~~~
D:\Project\open62541\open62541.c:192763:27: note: 'subjectURI' declared here
192763 |     UA_String             subjectURI;
       |                           ^~~~~~~~~~
D:\Project\open62541\open62541.c: At top level:
cc1.exe: note: unrecognized command-line option '-Wno-static-in-inline' may have been intended to silence earlier diagnostics
cc1.exe: all warnings being treated as errors
mingw32-make[2]: *** [CMakeFiles\open62541-object.dir\build.make:277: CMakeFiles/open62541-object.dir/open62541.c.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:441: CMakeFiles/open62541-object.dir/all] Error 2
mingw32-make: *** [Makefile:155: all] Error 2
```

Does somebody know how to fix this?

Thanks!

Greetings 

Mebus

kimim, 2021-11-23:
Hello, could you try 1.3 branch? There is no such error.

Mebus, 2021-11-23:
Hey Kimim,

I did a clean checkout of the 1.2.2 tag and built it with these options:

```
cmake -G "MinGW Makefiles" \
-DBUILD_SHARED_LIBS=ON \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DUA_NAMESPACE_ZERO=FULL \
-DCMAKE_INSTALL_PREFIX="D:\Project\opcuaclientexperiment\libopen62541" \
-D UA_ENABLE_AMALGAMATION=FALSE -DUA_ENABLE_ENCRYPTION=OPENSSL
```

Notice, that I removed the UA_ENABLE_AMALGAMATION=TRUE option. I think this resolved the issue.

Mebus

Mebus, 2021-11-24:
Hey folks,

with the CMake configuration in my first post I also get the following error on the `1.3` branch:

```
ntegrating file 'D:/Project/open62541-v1.3/plugins/crypto/ua_securitypolicy_none.c' ... done.
Integrating file 'D:/Project/open62541-v1.3/arch/win32/ua_clock.c' ... done.
Integrating file 'D:/Project/open62541-v1.3/arch/win32/ua_architecture_functions.c' ... done.
Integrating file 'D:/Project/open62541-v1.3/arch/network_tcp.c' ... done.
The size of D:/Project/open62541-v1.3/open62541.c is 10690050 Bytes.
[ 60%] Built target open62541-amalgamation-source
[ 60%] Built target open62541-code-generation
[ 65%] Building C object CMakeFiles/open62541-object.dir/open62541.c.obj
D:\Project\open62541-v1.3\open62541.c: In function 'namespace0_generated':
D:\Project\open62541-v1.3\open62541.c:173217:15: note: variable tracking size limit exceeded with '-fvar-tracking-assignments', retrying without
173217 | UA_StatusCode namespace0_generated(UA_Server *server) {
       |               ^~~~~~~~~~~~~~~~~~~~
D:\Project\open62541-v1.3\open62541.c: In function 'UA_CertificateVerification_VerifyApplicationURI':
D:\Project\open62541-v1.3\open62541.c:189332:9: error: 'subjectURI.length' may be used uninitialized [-Werror=maybe-uninitialized]
189332 |     if (UA_Bstrstr (subjectURI.data, subjectURI.length,
       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
189333 |                     applicationURI->data, applicationURI->length) == NULL) {
       |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Project\open62541-v1.3\open62541.c:189294:27: note: 'subjectURI' declared here
189294 |     UA_String             subjectURI;
       |                           ^~~~~~~~~~
D:\Project\open62541-v1.3\open62541.c:189332:31: error: 'subjectURI.data' may be used uninitialized [-Werror=maybe-uninitialized]
189332 |     if (UA_Bstrstr (subjectURI.data, subjectURI.length,
       |                     ~~~~~~~~~~^~~~~
D:\Project\open62541-v1.3\open62541.c:189294:27: note: 'subjectURI' declared here
189294 |     UA_String             subjectURI;
       |                           ^~~~~~~~~~
D:\Project\open62541-v1.3\open62541.c: At top level:
cc1.exe: note: unrecognized command-line option '-Wno-static-in-inline' may have been intended to silence earlier diagnostics
cc1.exe: all warnings being treated as errors
mingw32-make[2]: *** [CMakeFiles\open62541-object.dir\build.make:275: CMakeFiles/open62541-object.dir/open62541.c.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:441: CMakeFiles/open62541-object.dir/all] Error 2
mingw32-make: *** [Makefile:155: all] Error 2
```

Mebus

o6-nbeier, 2026-10-20:
Thank you for the report. The variable is initialized now (see #5821, #6009 and dc61a5a), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier