open62541/open62541 #4794: Extract String from LocalizedText

Psoke, 2021-11-23:
## Description
Hey there
I would like to store the value of a variable with the datatype LocalizedText locally for further porgramming.
ReadValueAttributes() stores the data in a UA_Variant:

```
typedef struct {
    const UA_DataType *type;     
    UA_VariantStorageType storageType;
    size_t arrayLength;       
    void *data;       
    size_t arrayDimensionsSize;
    UA_UInt32 *arrayDimensions;
} UA_Variant;
```
But encapsulated and just by reference via the data pointer.
The LocalizedText ist structured as following:

```
typedef struct {
    UA_String locale;
    UA_String text;
} UA_LocalizedText;
```

What is the most elegant way to get ahold of the string? My try was this:

```
UA_Variant Value;
UA_Variant_init(&Value);
retval = UA_Client_readValueAttribute(client, CurrentStateNodeId, &Value);

if (retval != UA_STATUSCODE_GOOD)
{
retval = PFC_CLIENT_STATUSCODE_CURRENTSTATE_ATTRIBUTE_ERROR;
}

if (Value.type == &UA_TYPES[UA_TYPES_LOCALIZEDTEXT])
{
UA_LocalizedText CurrentStateLocalizedText;
UA_LocalizedText_init(&CurrentStateLocalizedText);    //Do I need this?
UA_LocalizedText_copy(Value.data, CurrentStateLocalizedText)
UA_String CurrentStateString = UA_STRING(CurrentStateLocalizedText.text)
}
UA_Variant_clear(&Value);`
```
Copying it from the pointer to a local LocalizedText and extracting the string from there.
I tried to use something like this:

```
UA_String CurrentStateString = UA_STRING(Value.data->text)
```
But cant get it to work somehow. Any ideas?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-jvoe, 2021-11-23:
Just cast the data pointer of Value to UA_LocalizedText *. To be on the safe side, you should check first if the value is empty.

Psoke, 2021-11-24:
Just to be sure that I understood that correctly
Like so?

```
UA_LocalizedText CurrentStateLocalizedText = UA_LocalizedText *Value.data;
UA_String CurrentStateString = UA_STRING(&CurrentStateLocalizedText.text);
```

kimim, 2021-11-24:
@Psoke I think it is correct. Because `Value.data` contains all the bytes for `UA_LocalizedText`

basyskom-jvoe, 2021-11-24:
The .text member of UA_LocalizedText is already of type UA_String. The UA_STRING macro exists to make an UA_String from a null terminated char array, so it is unnecessary in this case.

jpfr, 2021-12-15:
Seems done here.

closed 2021-12-15 by jpfr