open62541/open62541 #4795: Error 'BadCommunicationError' was returned during OpenSecureChannel, FreeRTOS+LwIP

Max15421, 2021-11-23:
## Description
Hi everybody. I have a problem: I have a server runnning on my microcontroller. When I programm or reset my microcontroller and try to connect with a opc ua client I get the error: Error 'BadCommunicationError' was returned during OpenSecureChannel
The strange thing is that if I reset the controller severall times the connection gets established and everything works fine. I think thats somehow memory related but I don't understand whats the problem. Hope somebody can help me with this.

Cheers
Max

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2 
 - [x] Other OPC UA SDKs used (client or server): UA Expert 
 - [x] Operating system:  FreeRTOS + LWIP
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-12-14:
What does the open62541 debg log says or wireshark? Where did you get BadCommunicationError? You are sure the port is open?
The UaExpert report this error sometime if the listen socket from the server is not open.

Max15421, 2021-12-14:
I fix the problem. it was memory related. I was on the limit of my microcontroller. I deleted some nodes. now it works fine!

closed 2021-12-14 by Max15421