open62541/open62541 #4798: Question: Key generation for Basic256Sha256?

Mebus, 2021-11-24:
Hallo,

to use the Basic256Sha256 security module I generated my keys using the following command:

```
python open62541\tools\certs\create_self-signed.py -u urn:open62541.client.application -c client .
```

I extracted the server's key using UAExpert from the commercial server. 

Unfortuntely my output now looks like this:

```
Trying to connect to the server.
--------------------------------

[2021-11-24 16:06:53.041 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-11-24 16:06:53.041 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2021-11-24 16:06:53.041 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2021-11-24 16:06:53.041 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-11-24 16:06:53.041 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2021-11-24 16:06:53.042 (UTC+0100)] info/channel       Connection 384 | SecureChannel 2091015170 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-11-24 16:06:53.042 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-11-24 16:06:53.043 (UTC+0100)] info/client        Rejecting endpoint 0: security mode doesn't match
[2021-11-24 16:06:53.043 (UTC+0100)] info/client        Selected endpoint 1 in URL opc.tcp://MyPc:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-11-24 16:06:53.043 (UTC+0100)] info/client        Selected UserTokenPolicy Anonymous-[0]-SignAndEncrypt-Basic256Sha256 with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-11-24 16:06:53.043 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2021-11-24 16:06:53.043 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-11-24 16:06:53.043 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2021-11-24 16:06:53.043 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2021-11-24 16:06:53.043 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-11-24 16:06:53.044 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2021-11-24 16:06:53.045 (UTC+0100)] info/securitypolicy        The basic256sha256 security policy channel with openssl is created.
Assertion failed: keyLen == 256, file D:\Project\open62541\plugins\crypto\openssl\ua_openssl_basic256sha256.c, line 225
```

The key length seems to be incorrect. Did I generate the keys correctly with the command or is there a problem with the keys that I get form the server when I connect?

I am using open62541 version 1.2.2 at the moment.

Thanks

Greetings

Mebus

Mebus, 2021-11-24:
Hi folks,

this only seems to happen if I use the cmake option ` -DCMAKE_BUILD_TYPE=Debug ` instead of `-DCMAKE_BUILD_TYPE=RelWithDebInfo` is that an expected behaviour or a problem on my side? 🤔 

Mebus

ccvca, 2021-11-30:
Same as: #4545 ?
It looks like you created keys with a size other than 2048 (1024 or 4096).
As asserts only affect "Debug"-Builds, the error is gone in Release-Mode

jpfr, 2021-11-30:
The fix for 1.2 is in #4809. From there it will trickle down to 1.3 and master.
Thanks for reporting.

closed 2021-11-30 by jpfr