open62541/open62541 #4799: UA_Variant_setScaler sometimes creates an array of 1

Twoshrubs, 2021-11-25:
We have quite a mature open62541 project that has been running for the past year or so.. everything is going well apart from on the client side every now and then (about once in 4-5days eg. after millions of operations) they get a float array sent to them instead of a float.

The float array just contains the single float.

We have also seen the array in UAExpert Client logs and also in wireshark logs, as we checked just in case it was a client issue.

we use the following logic to populate the node..

UA_Variant_setScalar(&VarProcessValue, &processValue, &UA_TYPES[UA_TYPES_FLOAT]);
res = UA_Server_writeValue(myServer, UA_NODEID_NUMERIC(myNodeSpace, myNodeId_processVariable) , VarProcessValue);

I have added a few checks and data traps to hopefully stop sending this array to the client. 
So I was wondering if there something I was doing wrong or is there an issue in the 'setScalar'?

Were using Version 1.2.2 of Open62541 on Linux

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-11-25:
SetScalar is quite a primitive function.
```c
void
UA_Variant_setScalar(UA_Variant *v, void * UA_RESTRICT p,
                     const UA_DataType *type) {
    UA_Variant_init(v);
    v->type = type;
    v->arrayLength = 0;
    v->data = p;
}
```
Not much rooms for errors there.
So you run open62541 server-side and different clients?

Twoshrubs, 2021-11-26:
Hi Jpfr

Thank you for the help.

Yes, its a different client.. its an ABB client but we have also picked it up on UAExpert when data logging.
I added a trap yesterday so the code looks like this.

```
UA_Variant_setScalar(&VarProcessValue, &processValue, &UA_TYPES[UA_TYPES_FLOAT]);

 if(UA_Variant_isScalar(&VarProcessValue))
{
    res = UA_Server_writeValue(myServer, UA_NODEID_NUMERIC(myNodeSpace, myNodeId_processVariable) , VarProcessValue);
    if(res != 0)
    {
          LOGW << "Error trying to update the value of " << TagName.c_str() << " : Value = " << processValue << " : Result Code : " << res;
    }
}
else
{
    LOGD << "Array spotted";
}

```
I have to agree looking at the snippet you posted, i dont think that will be the issue.. We actually had an issue overnight and it wasnt captured in the log. So its making me think I'm doing something wrong with 'writeValue'.

Twoshrubs, 2021-11-26:
Just had some more info passed to me..

We seem to be getting the same data through twice on the client with a time difference between the packets but with the same server timestamps. 

I think I know what it is.. 
I did a change a few months ago to handle 1.#QNAN float values coming up from the PLC system and it used to lock up the node.. i think down to float compare issues. So I put in a 'isnan()' trap which I update the value as a 1.#QNAN, set the status but I don't update the timestamp.. (using UA_Server_Write)

For some reason it looks like for 0.0001% of the time the isnan() triggers for a float.. and it must be my logic in that trap which is causing the problem.

Thank you for your help, and sorry for raising this as an server issue.

closed 2021-11-26 by Twoshrubs