open62541/open62541 #4800: write callback fires on every read

deifemu, 2021-11-26:
## Description
I am trying to get a simple server running. I am close to your examples with the value callbacks. 
I am using the combined c and h files:  * Git-Revision: v1.2.2-dirty 
https://github.com/open62541/open62541/releases/download/v1.2.2/open62541.h
on windows with mingw

I am connecting to the server with UaExpert and everything looks fine except:
After every read from UaExpert a read and a write callback are fired. Even on read only nodes. 

What am I missing here? Did I do something wrong or is this a bug? 
I would like to have a write callback after a real write in UaExpert only. 

## Background Information / Reproduction Steps

	/* Add a variable node */
	/* 1) Define the node attributes */
	UA_VariableAttributes attr = UA_VariableAttributes_default;
	attr.displayName = UA_LOCALIZEDTEXT("en-US", name);
	UA_Variant_setScalar(&attr.value, &this->targetValue, &UA_TYPES[UA_TYPES_INT32]);
	if (rw){
		attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
	}

	/* 2) Define where the node shall be added with which browsename */
	UA_NodeId nodeId = UA_NODEID_STRING(1, name);
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_NodeId variableType = UA_NODEID_NULL; /* take the default variable type */
	UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, name);

	/* 3) Add the node */
	UA_Server_addVariableNode(this->server, nodeId, parentNodeId, parentReferenceNodeId,
							  browseName, variableType, attr, (void*)this, NULL);

	UA_ValueCallback callback ;
	callback.onRead = opcua_beforeRead;
	callback.onWrite = opcua_afterWrite;
	UA_Server_setVariableNode_valueCallback(this->server, nodeId, callback);

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-12-15:
We have two types of callbacks that can be attached to variables.
- datasource - where the data is stored in the backend
- value callback - where the data is stored in the node and updated before reading.

If you look at the example `tutorial_server_datasource.c`, the value is updated before every read.
The following code contains the culprit where a write is triggered before every read.
To update the value that is stored in the variable.

```c
static void
updateCurrentTime(UA_Server *server) {
    UA_DateTime now = UA_DateTime_now();
    UA_Variant value;
    UA_Variant_setScalar(&value, &now, &UA_TYPES[UA_TYPES_DATETIME]);
    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
    UA_Server_writeValue(server, currentNodeId, value); // <---  This line triggers the write callback during reading
}
```

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier